mavlink/qgroundcontrol #2308: Virtual Joystick issues

DonLakeFlyer, 2015-11-29:
- If you click/touch somewhere within the circle, the joystick jumps to that point. In real life, this would cause a radical change of control position. The position should only start to change once you hit the center of the joystick. Whatever point you started should be 0,0 (as the joystick is always auto centered when let go). At minimum, instead of jumping to the touched/clicked position, there should be a smooth ramp.
- There is a fixed high frequency timer (100Hz) sending joystick updates through the comm pipe. That timer should only be in high throttle when the stick is actually active (is 100Hz really necessary?). When idle (centered position), the timer could be a lot less frequent.
- Aesthetically, the joystick should not move beyond the circle. It is now moving all the way to the bounding rectangle.

DonLakeFlyer, 2015-11-29:
Also see #2307 for some discussion

pritamghanghas, 2015-12-23:
I would like to add one more. I need to try joystick first in actual flight. But I think this should be added.
Right now the stick dot moves back to zero if you thumb moves outside the circle. If I have correctly picked the joystick once as mentioned in point 1 of @DonLakeFlyer comment, it should get stuck at max position if the thumb is moving outside of bounding rectangle. Reason, we wouldn't be looking at screen while flying and thumb may move outside. Right now, I have to put my attention again on joystick and reacquire it.

closed 2016-01-18 by DonLakeFlyer