mavlink/qgroundcontrol #2327: Critical: REJ. WP CMD is showing in log

estechnical, 2015-12-01:
I built qgoundcontrol github master yesterday and tried it out yesterday and today. Today I noticed that I get "Critical: REJ. WP CMD: Partner ID Mismatch" showing up in the messages log just after it connects.

On one occasion of plugging in it showed a box warning of a critical safety issue and stated the same message that showed in the messages.

This doesn't happen with the previous version of qgroundcontrol I have installed (2.7.1 from the ubuntu ppa).

I checked my mavlink settings are all as per the px4 defaults for the PX4, is there anything else I should check?

Thanks,
Ed

DonLakeFlyer, 2015-12-02:
How are you connecting to the vehicle? Are you using anything in addition such as mavros or gazebo?

estechnical, 2015-12-02:
I'm connecting via mavlink using 3dr radios. I think the same message shows with USB connection, will also check that for you. Nothing else involved, just QGroundcontrol and PX4.

DonLakeFlyer, 2015-12-02:
Can you try to reproduce using master on both QGC and firmware side?

DonLakeFlyer, 2015-12-02:
Also does you vehicle have mission items on it already or not?

estechnical, 2015-12-02:
I'll have a look at loading the master firmware, can this be done with QGroundcontrol directly?

Just checked, the mission is empty (according to the old version of qgc) and the same message shows up when I try reading the mission info from the quad with master. FWIW I've never run a mission!

DonLakeFlyer, 2015-12-02:
> can this be done with QGroundcontrol directly?

Click the "Advanced" check box on firmware upgrade and you can select Master.

estechnical, 2015-12-02:
Just loaded master firmware to the PX4, lots of angry beeping later it booted up ok and now doesn't show this issue. For some reason I got lots of messages about data link 1 lost & regained on a couple of occasions. Now the original issue is gone and instead I have this showing up in my bootlog:

    PX4IO Trying to update 
    ERROR: PX4IO update failed 
    ERROR: PX4IO not found 
    ERROR: Could not load mixer: /etc/mixers/quad_x.main.mix 
    PX4IO Trying to update 
    ERROR: PX4IO update failed 
    ERROR: PX4IO not found 
    ERROR: Could not load mixer: /etc/mixers/quad_x.main.mix 
    PX4IO Trying to update 
    ERROR: PX4IO update failed 
    ERROR: PX4IO not found 
    PX4IO Trying to update 
    ERROR: PX4IO update failed 
    ERROR: PX4IO not found 
    ERROR: Could not load mixer: /etc/mixers/quad_x.main.mix 
    PX4IO Trying to update 
    ERROR: PX4IO update failed 
    ERROR: PX4IO not found 
    ERROR: Could not load mixer: /etc/mixers/quad_x.main.mix

All is not well with master firmware and QGC seems to show everything being ok.

I should go back to a stable firmware now...

LorenzMeier, 2015-12-10:
This looks like a specific problem for your setup, potentially including a HW fault. Closing.

closed 2015-12-10 by LorenzMeier