mavlink/qgroundcontrol #2328: Segmentation fault on startup

estechnical, 2015-12-01:
Sometimes when starting latest master I get a segmentation fault as soon as the GUI appears. Here is the output:
    ./qgroundcontrol-start.sh 
    Logging ini file directory "/home/ed/.config/QtProject"
    Filter rules "qt.network.ssl.warning=false\n"
    Settings location "/home/ed/.config/QGroundControl.org/QGroundControl.ini" true
    Saved files location "/home/ed/Documents/QGroundControl"
    Mapping cache directory: "/home/ed/.cache/QGCMapCache55"
    0
    Orphaned log file count 0
    Segmentation fault (core dumped)

I'm using ubuntu 14.04 and since I saw the other issue about segfault on start up, I restarted and checked again to be sure. It still happens sometimes.

DonLakeFlyer, 2015-12-02:
Have you ever run qgroundcontrol using sudo?

estechnical, 2015-12-02:
Not that I can recall - I'll check my shell history when I get the laptop out... But I have alternated use of the PPA version and the new version built from source.

DonLakeFlyer, 2015-12-16:
Can you try latest master?

closed 2015-12-16 by DonLakeFlyer

estechnical, 2015-12-18:
Just tested - this is now working reliably every time I start using either the run button from QTcreator or the qgroundcontrol-start.sh. Weirdly it did the same old segfault just the first time I tried it, then has behaved correctly on the subsequent 5 tries...