mavlink/qgroundcontrol #2334: Custom command widget shows "no vehicle connected" even when a vehicle is connected

mairsbw, 2015-12-03:
Note that this is for my own autopilot, which doesn't even exist in the MAVLink `MAV_AUTOPILOT` enum. What do custom autopilots need to do to make QGC work properly? It used to be enough to just transmit `SYS_STATUS` and `HEARTBEAT` to QGC and this functionality worked. Is there something that changed in the move to QML-based widgets?

DonLakeFlyer, 2015-12-03:
Probably just a bug in custom command widget. Lots of changes recently with respect to vehicles and connections.

DonLakeFlyer, 2015-12-03:
Unrelated to this bug: If you have a custom autopilot just make sure it doesn't report itself with the PX4 or APM MAV_AUTOPILOT value.

DonLakeFlyer, 2015-12-03:
CustomCommand widget waits for parameter load to complete before it makes the vehicle available to it. So your autopilot needs to responds to the param request (even if it has none). One param load is complete, it should switch away from the "No vehicle available".

DonLakeFlyer, 2015-12-12:
@mairsbw Any news on this one?

mairsbw, 2015-12-15:
Yep, we've got it working. It's pretty unreliable though, but I'm fairly certain the problem is with our parameter protocol state machine.

closed 2015-12-15 by mairsbw