mavlink/qgroundcontrol #2335: App crash on exit with serial link connected

LorenzMeier, 2015-12-03:
```
Process:               qgroundcontrol [76305]
Path:                  /Users/USER/*/qgroundcontrol.app/Contents/MacOS/qgroundcontrol
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               2.0.0 (227)
Code Type:             X86-64 (Native)
Parent Process:        Qt Creator [72689]
Responsible:           Qt Creator [72689]
User ID:               502

Date/Time:             2015-12-03 14:48:07.683 +0100
OS Version:            Mac OS X 10.10.5 (14F27)
Report Version:        11
Anonymous UUID:        78D96205-41B6-10DA-A0C7-BE314875235A

Sleep/Wake UUID:       7EA1E4B2-351B-455C-9237-80F6CA5E23A8

Time Awake Since Boot: 1700000 seconds
Time Since Wake:       4600 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000005458

VM Regions Near 0x5458:
--> 
    __TEXT                 0000000100000000-0000000100667000 [ 6556K] r-x/rwx SM=COW  /Users/USER/*/qgroundcontrol.app/Contents/MacOS/qgroundcontrol

Application Specific Information:
Performing @selector(_close:) from sender _NSThemeCloseWidget 0x10b00c5b0

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtQml          	0x0000000101b94c16 QQmlData::isSignalConnected(QAbstractDeclarativeData*, QObject const*, int) + 22
1   org.qt-project.QtCore         	0x0000000100be9799 QMetaObject::activate(QObject*, int, int, void**) + 105
2   org.qgroundcontrol.qgroundcontrol	0x0000000100247520 LinkConfiguration::linkChanged(LinkInterface*) + 64
3   org.qgroundcontrol.qgroundcontrol	0x0000000100169dda SerialLink::~SerialLink() + 42
4   org.qgroundcontrol.qgroundcontrol	0x0000000100169f7f SerialLink::~SerialLink() + 15
5   org.qgroundcontrol.qgroundcontrol	0x00000001000aebaf LinkManager::_deleteLink(LinkInterface*) + 335
6   org.qgroundcontrol.qgroundcontrol	0x00000001000ae739 LinkManager::disconnectLink(LinkInterface*) + 153
7   org.qgroundcontrol.qgroundcontrol	0x00000001000b1701 LinkManager::shutdown() + 177
8   org.qgroundcontrol.qgroundcontrol	0x000000010014115a MainWindow::closeEvent(QCloseEvent*) + 266
9   org.qt-project.QtWidgets      	0x000000010209122c QWidget::event(QEvent*) + 2236
10  org.qt-project.QtWidgets      	0x000000010219b741 QMainWindow::event(QEvent*) + 929
11  org.qt-project.QtWidgets      	0x00000001020524bb QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
12  org.qt-project.QtWidgets      	0x000000010205589e QApplication::notify(QObject*, QEvent*) + 8318
13  org.qt-project.QtCore         	0x0000000100bb9983 QCoreApplication::notifyInternal(QObject*, QEvent*) + 115
14  org.qt-project.QtWidgets      	0x00000001020809ed QWidgetPrivate::close_helper(QWidgetPrivate::CloseMode) + 189
15  org.qt-project.QtWidgets      	0x00000001020b0aa1 QWidgetWindow::event(QEvent*) + 769
16  org.qt-project.QtWidgets      	0x00000001020524bb QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
17  org.qt-project.QtWidgets      	0x000000010205589e QApplication::notify(QObject*, QEvent*) + 8318
18  org.qt-project.QtCore         	0x0000000100bb9983 QCoreApplication::notifyInternal(QObject*, QEvent*) + 115
19  org.qt-project.QtGui          	0x00000001014451fd QGuiApplicationPrivate::processCloseEvent(QWindowSystemInterfacePrivate::CloseEvent*) + 109
20  org.qt-project.QtGui          	0x0000000101440f9d QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 541
21  org.qt-project.QtGui          	0x000000010142d9da QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 346
22  org.qt-project.QtGui          	0x000000010142b9a1 QWindowSystemInterface::flushWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 673
23  libqcocoa.dylib               	0x0000000106fe4dfe QCocoaWindow::windowShouldClose() + 46
24  libqcocoa.dylib               	0x0000000106ff207b -[QNSWindowDelegate windowShouldClose:] + 27
25  com.apple.AppKit              	0x00007fff91a6aa63 __19-[NSWindow __close]_block_invoke + 134
26  com.apple.AppKit              	0x00007fff91a6a9d0 -[NSWindow __close] + 312
27  libsystem_trace.dylib         	0x00007fff97e04cd7 _os_activity_initiate + 75
28  com.apple.AppKit              	0x00007fff91945eb1 -[NSApplication sendAction:to:from:] + 452
29  com.apple.AppKit              	0x00007fff9195b946 -[NSControl sendAction:to:] + 86
30  com.apple.AppKit              	0x00007fff9195b862 __26-[NSCell _sendActionFrom:]_block_invoke + 131
31  libsystem_trace.dylib         	0x00007fff97e04cd7 _os_activity_initiate + 75
32  com.apple.AppKit              	0x00007fff9195b7bf -[NSCell _sendActionFrom:] + 144
33  libsystem_trace.dylib         	0x00007fff97e04cd7 _os_activity_initiate + 75
34  com.apple.AppKit              	0x00007fff91959cb3 -[NSCell trackMouse:inRect:ofView:untilMouseUp:] + 2821
35  com.apple.AppKit              	0x00007fff919b234f -[NSButtonCell trackMouse:inRect:ofView:untilMouseUp:] + 770
36  com.apple.AppKit              	0x00007fff91958366 -[NSControl mouseDown:] + 714
[... 176 lines cut ...]

Thread 18:

Thread 19:: QScanThread
0   libsystem_kernel.dylib        	0x00007fff9a79951a semaphore_wait_trap + 10
1   libdispatch.dylib             	0x00007fff97b4ac55 _dispatch_semaphore_wait_slow + 213
2   com.apple.framework.CoreWLAN  	0x00007fff99a29755 -[CWInterface(Private) scanForNetworksWithChannels:ssidList:legacyScanSSID:includeHiddenNetworks:mergedScanResults:maxAge:maxMissCount:maxWakeCount:maxAutoJoinCount:waitForWiFi:waitForBluetooth:priority:error:] + 638
3   com.apple.framework.CoreWLAN  	0x00007fff99a23fdc -[CWInterface scanForNetworksWithName:error:] + 109
4   libqcorewlanbearer.dylib      	0x000000010c8948ba QScanThread::run() + 218
5   org.qt-project.QtCore         	0x0000000100a0ed93 QThreadPrivate::start(void*) + 339
6   libsystem_pthread.dylib       	0x00007fff976a305a _pthread_body + 131
7   libsystem_pthread.dylib       	0x00007fff976a2fd7 _pthread_start + 176
8   libsystem_pthread.dylib       	0x00007fff976a03ed thread_start + 13

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000005458  rbx: 0x00007fff7b252070  rcx: 0x000000010c26e060  rdx: 0x0000000000000004
  rdi: 0x0000000100000064  rsi: 0x2000000000000004  rbp: 0x00007fff5fbfda70  rsp: 0x00007fff5fbfda70
   r8: 0x0000000101b94c00   r9: 0x0000000110400000  r10: 0x0000000000000040  r11: 0x0000000110400000
  r12: 0x0000000000000004  r13: 0x0000000102cbb1d0  r14: 0x00007fff5fbfdbc0  r15: 0x000000010ef9d9a0
  rip: 0x0000000101b94c16  rfl: 0x0000000000010202  cr2: 0x0000000000005458
  
Logical CPU:     2
Error Code:      0x00000004
Trap Number:     14
```

dogmaphobic, 2015-12-03:
I've been seeing some odd behaviors here and there but I haven't seen any crash. Where were you when you exited it? The stack trace above shows the serial link being destroyed and sending a signal notifying about it. It looks like the signal was making its way up to the QML side when the app crashed.

The source of the signal:

https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/LinkConfiguration.cc#L157

LorenzMeier, 2015-12-03:
Flight view. Its on my MacBook air, so it could be that its slower than the machines we're using. Could also explain some Android crashes I've seen.

DonLakeFlyer, 2015-12-03:
Do you have QT_FATAL_WARNINGS set as an environment variable?

LorenzMeier, 2015-12-04:
No I don't.

DonLakeFlyer, 2015-12-12:
I think I may have just fixed this. Can you try again?

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier