mavlink/qgroundcontrol #2336: Autoconnect: QGC does not auto-connect after successful Firmware Upgrade

LorenzMeier, 2015-12-03:
As the title says. It just sits there and shows nothing, which is very confusing from an user perspective.

LorenzMeier, 2015-12-03:
I also switched to the summary perspective if it was just locking on the firmware upgrade dialog. But even then I think it would be better to auto-connect and push the user to the summary screen, as he will not want to upgrade Firmware again.

DonLakeFlyer, 2015-12-03:
Is this on latest master? Just tried it an it worked fine.

LorenzMeier, 2015-12-04:
Yes, latest master the moment I reported it. I can re-test.

DonLakeFlyer, 2015-12-04:
Maybe that was when auto connect was really slow? Don't quite understand. Can you retry again? If it doesn't work. Run with --logging:LinkManagerLog,MultiVehicleManagerLog so I can try to debug remotely.

DonLakeFlyer, 2015-12-06:
I've hit this intermittently, but still not sure why.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier