mavlink/qgroundcontrol #2338: iOS MessageDialog and Text Field issues

dogmaphobic, 2015-12-03:
Putting here so it doesn't get lost:

The *MessageDialog* QML Item does not work on iOS.
The *TextField* (or our derived QGCTextField) does not bring up the keyboard for you to enter text.

Given that these seem to work on pure QML code, my only assumption is that it has something to do with QtQuick inside QtWidget.

These two are a show stopper for iOS.

DonLakeFlyer, 2015-12-03:
> The TextField (or our derived QGCTextField) does not bring up the keyboard for you to enter text.

Most likely needs additional hinting like this:
https://github.com/mavlink/qgroundcontrol/blob/master/src/FactSystem/FactControls/FactTextField.qml#L21

dogmaphobic, 2015-12-03:
I tried using TextField straight (not QGCTextField) with and without hints. It all works on Android. Just not on iOS.

dogmaphobic, 2015-12-03:
The example code, which is a pure QML application, runs fine. It might look weird (not a system dialog) but it works. That's what makes me believe it's an issue with QtQuick inside QtWidgets. My guess is that we won't have a viable iOS app until we are fully away from QtWidgets.

[image]

[image]

[image]

DonLakeFlyer, 2015-12-03:
I’d enter a bug. Maybe we can get a fix in next Qt. That may happen before we get rid of QQuickWidget.

dogmaphobic, 2015-12-09:
Well, moving away from QtWidgets fixed it all :)

closed 2015-12-09 by dogmaphobic