mavlink/qgroundcontrol #2339: Showstopper: Waypoint editor sets an acceptance radius of zero

LorenzMeier, 2015-12-03:
@DonLakeFlyer The current waypoint editor sets and acceptance radius of zero for mission items, which makes them impossible to reach. The radius should default to 3m for multicopters and ~20m for planes. Users might need control over this.

Flying missions is completely broken with this.

thomasgubler, 2015-12-03:
+1 this confused me as well since I expected a reasonable default value

DonLakeFlyer, 2015-12-03:
Stable? Master? Both?

thomasgubler, 2015-12-03:
master this morning 71c419621e785cfdb80f32fccc6f0a7d3f62a91a

DonLakeFlyer, 2015-12-03:
Master only? Can someone help with testing Stable?

thomasgubler, 2015-12-03:
the issue is still present on current master, going to build stable now

LorenzMeier, 2015-12-03:
I pushed a workaround to PX4/Firmware master which uses the default radius is the one from the GCS is very small.

thomasgubler, 2015-12-03:
stable seems ok

DonLakeFlyer, 2015-12-03:
> The radius should default to 3m for multicopters and ~20m for planes.

There is no default different between Multi and Fixed it's 3m for both. Have mission editing understand aircraft type is tricky since you don't know it when you are offline. If this is critical put in a separate issue on it.

DonLakeFlyer, 2015-12-03:
I'd vote for moving acceptance radius to a user visible friendly edit param.

closed 2015-12-03 by DonLakeFlyer

dagar, 2015-12-03:
Shouldn't this default to NAV_ACC_RAD?
https://pixhawk.org/firmware/parameters#mission

DonLakeFlyer, 2015-12-03:
Don't know. Same thing with NAV_LOITER_RAD. Those take over if some value is set. Maybe 0? Doesn't say. @LorenzMeier ?

LorenzMeier, 2015-12-04:
Do, you set this in the mission editing. I've now changed this so anything non-zero sets it and zero sets the default.

DonLakeFlyer, 2015-12-04:
> I've now changed this so anything non-zero sets it

Not sure what that means. Do you mean non-zero acceptance radius and firmware uses the value aa is?