mavlink/qgroundcontrol #2371: New daily build is crashing afrer several seconds of run

Fojtik, 2015-12-08:
New daily build is crashing after several seconds of run.

The log from my emulator is attached. There seems nothing to be wrong:
[DTP2MAVLINK000.txt](https://github.com/mavlink/qgroundcontrol/files/55668/DTP2MAVLINK000.txt)

This is an output of qgroundcontrol.exe:
D:\Program Files\qgroundcontrol>qgroundcontrol.exe --full-logging  1>full_log.txt
[I] at :0 - "Logging ini file directory "H:/Documents and Settings/Administrator.DVRMN/Local Settings/Data aplikac├ş/QtProject""
[I] at :0 - "Filter rules "qt.network.ssl.warning=false\n""
[I] at :0 - "Settings location "H:/Documents and Settings/Administrator.DVRMN/Data aplikac├ş/QGroundControl.org/QGroundControl.ini" true"
[I] at :0 - "Saved files location "H:/Documents and Settings/Administrator.DVRMN/Dokumenty/QGroundControl""
[I] at :0 - "Mapping cache directory: "H:/Documents and Settings/Administrator.DVRMN/Local Settings/Data aplikac├ş/cache/QGCMapCache55""
[I] at :0 - "0"
[I] at :0 - "Orphaned log file count 0"
[I] at :0 - "Temp log "H:/Documents and Settings/Administrator.DVRMN/Local Settings/Temp/FlightData232216.mavlink" false"

Fojtik, 2015-12-08:
And this is a crash callstack:
>	qgroundcontrol.exe!MissionItem::friendlyEditAllowed()  Line 674 + 0x6b bytes	C++
 	qgroundcontrol.exe!MissionItem::_sendFriendlyEditAllowedChanged()  Line 834 + 0x5 bytes	C++
 	Qt5Core.dll!67189afc() 	
 	[Frames below may be incorrect and/or missing, no symbols loaded for Qt5Core.dll]	
 	Qt5Core.dll!67189dae() 	
 	qgroundcontrol.exe!Fact::valueChanged(QVariant _t1={...})  Line 308 + 0x1b bytes	C++
 	qgroundcontrol.exe!Fact::setRawValue(const QVariant & value={...})  Line 103	C++
 	qgroundcontrol.exe!MissionItem::setCommand(MAV_CMD command=0x00000000)  Line 426 + 0x23 bytes	C++
 	qgroundcontrol.exe!MissionItem::MissionItem(int sequenceNumber=0x00000000, MAV_CMD command=0x00000000, MAV_FRAME frame=MAV_FRAME_GLOBAL, double param1=0.00000000000000000, double param2=0.00000000000000000, double param3=80.000000000000000, double param4=0.00000000000000000, double param5=50.361232757568359, double param6=13.227539062500000, double param7=505.00000000000000, bool autoContinue=false, bool isCurrentItem=true, QObject * parent=0x06479f78)  Line 179	C++
 	qgroundcontrol.exe!MissionManager::_handleMissionItem(const __mavlink_message & message={...})  Line 301 + 0xd0 bytes	C++
 	qgroundcontrol.exe!QtPrivate::QSlotObject<void (__thiscall MissionManager::*)(__mavlink_message const &),QtPrivate::List<__mavlink_message const &>,void>::impl(int which=0x00000001, QtPrivate::QSlotObjectBase * this_=0x0647b0d0, QObject * r=0x06479f78, void * * a=0x0013d240, bool * ret=0x00000000)  Line 150	C++
 	Qt5Core.dll!67189afc() 	
 	Qt5Core.dll!67189dae() 	
 	qgroundcontrol.exe!Vehicle::mavlinkMessageReceived(const __mavlink_message & _t1={...})  Line 1124	C++
 	qgroundcontrol.exe!Vehicle::_mavlinkMessageReceived(LinkInterface * link=0x063e7a28, __mavlink_message message={...})  Line 230	C++
 	qgroundcontrol.exe!QtPrivate::QSlotObject<void (__thiscall Vehicle::*)(LinkInterface *,__mavlink_message),QtPrivate::List<LinkInterface *,__mavlink_message>,void>::impl(int which=0x00000001, QtPrivate::QSlotObjectBase * this_=0x06471e10, QObject * r=0x063ed6c0, void * * a=0x0013d660, bool * ret=0x00000000)  Line 150	C++
 	Qt5Core.dll!67189afc() 	
 	Qt5Core.dll!67189dae() 	
 	qgroundcontrol.exe!MAVLinkProtocol::messageReceived(LinkInterface * _t1=0x063e7a28, __mavlink_message _t2={...})  Line 469	C++
 	qgroundcontrol.exe!MAVLinkProtocol::receiveBytes(LinkInterface * link=0x063e7a28, QByteArray b={...})  Line 416	C++
 	qgroundcontrol.exe!QtPrivate::QSlotObject<void (__thiscall MAVLinkProtocol::*)(LinkInterface *,QByteArray),QtPrivate::List<LinkInterface *,QByteArray>,void>::impl(int which=0x00000001, QtPrivate::QSlotObjectBase * this_=0x06467d68, QObject * r=0x00193008, void * * a=0x06451458, bool * ret=0x00000000)  Line 150	C++
 	Qt5Core.dll!6718e0eb() 	
 	Qt5Core.dll!6718d23c() 	
 	Qt5Core.dll!6723853b() 	
 	Qt5Widgets.dll!65012779() 	
 	Qt5Widgets.dll!65011750() 	
 	Qt5Gui.dll!01002db4() 	
 	Qt5Gui.dll!00e6114a() 	
 	Qt5Gui.dll!00e631eb() 	
 	Qt5Gui.dll!00e62e86() 	
 	Qt5Gui.dll!00e62cf6() 	
 	qwindows.dll!01aca538() 	
 	Qt5Core.dll!6716abfb() 	
 	SSSensor.dll!061f265f() 	
 	msvcr120.dll!00afecfa() 	
 	Qt5Core.dll!6716abfb() 	
 	Qt5Core.dll!6716c377() 	
 	Qt5Core.dll!671ac29f() 	
 	qwindows.dll!01acbd29() 	
 	Qt5Core.dll!671ab8f6() 	
 	user32.dll!7e37b372() 	
 	user32.dll!7e368734() 	
 	user32.dll!7e368816() 	
 	user32.dll!7e3689cd() 	
 	Qt5Core.dll!671671a8() 	
 	Qt5Core.dll!671ab196() 	
 	atioglxx.dll!69ec4d25() 	
 	qgroundcontrol.exe!MAVLinkDecoder::MAVLinkDecoder(MAVLinkProtocol * protocol=0xce8b0c76, QObject * parent=0x1e6c15ff)  Line 13 + 0xcf39 bytes	C++
 	ff046a00()

DonLakeFlyer, 2015-12-10:
Can you capture a telemetry log in QGC and attach here so I can play it back? Turn on logging even when not armed so you'll get prompted to save when you disconnect.
[image]

Fojtik, 2015-12-10:
Dear Don,

> Can you capture a telemetry log in QGC and attach here so I can play it back? Turn on logging even when not armed so you'll get prompted to save when you disconnect.
> [image]
Please try attached file to this Email.
And report me whether you succeed.

regards
  Jara

DonLakeFlyer, 2015-12-10:
You can't attach files to a github email reponse. They don't come through. Just email the telemetry log directly to me at don@thegagnes.com

DonLakeFlyer, 2015-12-12:
The crash is most likely because you are using a MAV_CMD id which doesn't exist. If I remember correctly you are using a MAV_CMD=0 somewhere in your mission. Instead of using a non-supported MAV_CMD (one that is not in the mavlink headers), you should re-use one that does exist in the headers. Then it won't crash.

Fojtik, 2015-12-12:
Even if so, the QGC MUST NEVER crash due to invalid data on input! It might data reject, but not crash.

I do not see MAV_CMD_USER in the enum  MAV_CMD:
https://github.com/mavlink/c_library/blob/master/common/common.h
   Please help me which value to choose.

DonLakeFlyer, 2015-12-12:
> Even if so, the QGC MUST NEVER crash due to invalid data on input!

Correct, but until that is fixed this is a workaround. I'm still trying to figure out that best way to handle an unknown MAV_CMD. MAV_CMD_USER has not yet been added to mavlink. Until then just use something like MAV_CMD_NAV_FOLLOW. I'll put in the pull to mavlink today, then fix QGC crash in the next coupe days. Up to you as to what you want to wait for.

DonLakeFlyer, 2015-12-12:
Mavlink pull is here: https://github.com/mavlink/mavlink/pull/480

DonLakeFlyer, 2015-12-16:
Crash is fixed.

closed 2015-12-16 by DonLakeFlyer

Fojtik, 2015-12-18:
Yes, you are right, the crash is no longer observed ;).

DonLakeFlyer, 2015-12-18:
Great. Thanks for reporting back.