mavlink/qgroundcontrol #2387: Param loading in PX4 hosed when connected via serial

LorenzMeier, 2015-12-13:
@DonLakeflyer could you try this with current master in your setup? @pkocmoud has issues with latest daily, but not with one a couple days old. Note that his issues also vary between OS. You can reach him via Gitter.

DonLakeFlyer, 2015-12-13:
Connected via serial: Serial what? USB or Radio?

pkocmoud, 2015-12-13:
Hi Don, 
My Linux computer has a much more recent version on it. I am using /dev/ttyS0 on the PixRacer and I have the same issue with either the 8266 TCP-Serial bridge and a FTDI cable. Linux v2.8.0-202 does not work / PC v2.8.0-154 Works.

DonLakeFlyer, 2015-12-13:
Can you run QGC with --logging:ParameterLoaderLog and post the output here?

pkocmoud, 2015-12-13:
I am connected, I see HUD activity but I have no setup options including Parameters
On the command line I see:

ParameterLoaderLog: Request to refresh all parameters

That is the last line.

Can we chat via Skype?

pkocmoud, 2015-12-13:
I emailed you Screen shots. I think QGC is conflicted as to if it is connected or not. USB connections works fine.

DonLakeFlyer, 2015-12-13:
If parameters don't come across you won't get full ui.

DonLakeFlyer, 2015-12-15:
Run with new master shows that it is likely that Vehicle is not responding to PARAM_REQUEST_LIST.

DonLakeFlyer, 2015-12-21:
Is this a real problem on the QGC side? Doesn't seem like it.

pkocmoud, 2015-12-21:
This seems to be resolved with Gus's new 8266 firmware.

closed 2015-12-21 by DonLakeFlyer