mavlink/qgroundcontrol #2390: APM flash needs to show versions instead of Stable/Beta/Master...

DonLakeFlyer, 2015-12-13:
This is what APM users are familiar with

estechnical, 2015-12-18:
It would be nice if the version was shown for PX4 stack too - personally I'm never sure what I am loading since stable/beta/master are just names and could change at any time.

Since the version is not shown, how to know whether there is a stable release that's newer than my current stable firmware?

DonLakeFlyer, 2015-12-18:
I'll look at that. There didn't used to be real version information in the past that I could get to for PX4 Stack. @LorenzMeier ?

LorenzMeier, 2015-12-18:
I use git tags, like here: https://github.com/PX4/Firmware/releases

I think we should be using the Github API to pull down the release JSON and fill the drop down using it:
https://developer.github.com/v3/repos/releases/

Here is the JSON for it:
https://api.github.com/repos/PX4/Firmware/releases

DonLakeFlyer, 2015-12-18:
But how to determine Stable versus Beta?

LorenzMeier, 2015-12-18:
See the pre-release: true / false flag.

LorenzMeier, 2015-12-26:
@DonLakeFlyer What about the release info for PX4? Given you have the infrastructure now, it would make sense to have it in as well.

tubeme, 2016-01-28:
I was never sure what I was loading until I started manually downloading all the stable/beta and masters, mark them and keep track of versions I flash myself. There is even harder problem. If I want to see what I've loaded in the FC let say three months ago on a certain machine. We fly couple of machines and couple of FC's at the same time.

DonLakeFlyer, 2016-02-13:
See Issue #2797

DonLakeFlyer, 2016-05-29:
Just showing Stable version for 3.0. Rest of the work is covered in #2797 which is moving to 3.1

closed 2016-05-29 by DonLakeFlyer