mavlink/qgroundcontrol #2394: add PX4 geofence params to setup

dagar, 2015-12-14:
I'd like to add the PX4 geofence parameters to the QGC setup. At a minimum we should expose the action and max horizontal and vertical distance params.

GF_ACTION
0 = none, 1 = warning (default), 2 = loiter, 3 = return to launch, 4 = fight termination

GF_MAX_HOR_DIST
Max horizontal distance in meters

GF_MAX_VER_DIST
Max vertical distance in meters

Would it be difficult to display GF_MAX_HOR_DIST as a circle on the map?

https://pixhawk.org/firmware/parameters#geofence

LorenzMeier, 2015-12-15:
@DonLakeFlyer Could we get this in? We're nearing a release and it would be fantastic to have this covered.

DonLakeFlyer, 2015-12-16:
The geo fence params need good firmware meta data associated with them. Units at a minimum.

dagar, 2015-12-16:
Geofence units and min added - https://github.com/PX4/Firmware/pull/3364

closed 2015-12-16 by DonLakeFlyer