mavlink/qgroundcontrol #2398: escape dpesm

pritamghanghas, 2015-12-14:


pritamghanghas, 2015-12-14:
[image]
[image]

DonLakeFlyer, 2015-12-14:
Parameter editor dialog has Escape key handling:
```
            Keys.onReleased: {
                if (event.key == Qt.Key_Escape) {
                    reject()
                }
            }
```

Seems to work on OSX. Maybe an Ubuntu only problem.

pritamghanghas, 2015-12-14:
ok, I will have a look why its not working on Ubuntu, being largely on
Ubuntu will make it easier for me.

pritamghanghas, 2015-12-14:
ok some debugging results. The text field never gets focus on its own. If I
click on it to give it focus and then press escape, it correctly goes away
but the window still minimizes but the screen remains usable. If I invoke
text field correctly gains focus. Does that trigger any ideas.

DonLakeFlyer, 2015-12-14:
I pulled out moving the focus to the text field a while back because it was causing other problems. Likely the whole handling of Escape needs to be reworked. I'd just pull it out for now. And then add an issue which says to add Escape/Enter and focus support for dialogs. Then we can figure out the right way to do this.

pritamghanghas, 2015-12-14:
I will dig at it for a while tomorrow before pulling it out. you pulled out
only forcing the focus there. The property focus: true is still there and
somehow it is getting focus correctly once I have clicked on it even if
invoked again. Something else going on.

pritamghanghas, 2015-12-23:
this is partly fixed by #2425. Changed the title accordingly.

DonLakeFlyer, 2015-12-25:
I have this almost fixed now. Need to do some more testing as well as double check would Qml Dialog control is implemented.

pritamghanghas, 2015-12-26:
great. I would love to see how did you fixed it. My knowledge of qml is not
even at beginner level. I read through docs but didn't really work.

closed 2015-12-27 by DonLakeFlyer