mavlink/qgroundcontrol #2431: Download files from the pixhawk through the qgc's Onboard Files

CornerOfSkyline, 2015-12-20:
I found it is hard to download the log file from pixhawk through the QGC Onboardfile . When the log file's size larger than 10MB, the download will break with "Bad sequence". Can we improve the download method? Adding some handshake or through request and send to support continue to download file even have some bad sequence.

DonLakeFlyer, 2015-12-20:
I assume you are connected via USB, correct? File download is not a robust protocol which can handle lost packets. It will only work over USB. If you are getting a " bad sequence" error then there is a bug somewhere either in Firmware or QGC since you shouldn't lose packets over USB.

LorenzMeier, 2015-12-20:
@DonLakeFlyer There is no fundamental reason it can't be robust. And I'm seeing these even over USB, so I suspect the root cause being actually an implementation and / or protocol bug, potentially some sort of race or bad timeout.

DonLakeFlyer, 2015-12-20:
> There is no fundamental reason it can't be robust.

The fundamental reason is that it would take a lot of work!

dogmaphobic, 2015-12-20:
Just a guess. If it's a *bad sequence*, could it be a case of a counter overflow and starting back at 0? Or worse, a negative number for a badly typed signed int type?

dogmaphobic, 2015-12-20:
That is, it appears these errors are for large files, which would indicate a very large number of *packets* (I don't know the packet size). If it's a small uint, it may be overflowing. If one end treats the counter as int and the other as uint, you would get negative numbers once it overflows.

DonLakeFlyer, 2015-12-20:
Yup, that's probably it. Sequence number is only 16 bits so it will roll over on larger files. Code likely doesn't handle roll-over correctly.

dogmaphobic, 2015-12-20:
It appears both sides use uint16_t:
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_ftp.h#L73
https://github.com/mavlink/qgroundcontrol/blob/master/src/uas/FileManager.h#L208

dogmaphobic, 2015-12-20:
> Yup, that's probably it. Sequence number is only 16 bits so it will roll over on larger files. Code likely doesn't handle roll-over correctly.

You beat me to it :)

dogmaphobic, 2015-12-20:
Yep, this code should take overflow into account:

https://github.com/mavlink/qgroundcontrol/blob/master/src/uas/FileManager.cc#L337-L339

dogmaphobic, 2015-12-20:
Firmware side has no check for sequence number. It just keep incrementing it.

dogmaphobic, 2015-12-20:
Sorry for spamming this :) 

Actually, the current code should overflow accordingly. That is, it would be comparing 0 to 0 when it goes beyond 0xFFFF:

```
 uint16_t incomingSeqNumber = request->hdr.seqNumber;
    
    // Make sure we have a good sequence number
    uint16_t expectedSeqNumber = _lastOutgoingSeqNumber + 1;
    if (incomingSeqNumber != expectedSeqNumber) {
```

DonLakeFlyer, 2015-12-20:
Yeah I know. That may not be what is going on here. Visual inspection seem to show that overflow should just work correctly. I'll have to do it the hard way with a real board!

dogmaphobic, 2015-12-20:
Ugh... Just set both sides to initialize the counter to 0x0FFF0 so you can get to it quicker :)

DonLakeFlyer, 2015-12-20:
It's not rollover:
[image]

dogmaphobic, 2015-12-20:
40 packets?

DonLakeFlyer, 2015-12-20:
@LorenzMeier FYI: Looks like a firmware side problem so far. Wanted to let you know since you are working towards release there. I may not have time time to debug firmware side to way later tonight or maybe tomorrow. Looks like a bug in burst mode streaming sequence number update.

dogmaphobic, 2016-01-03:
This needs a bit of an architectural change on the client side (QGC). As it stands, once a bad sequence is detected, the transfer is flagged as failed and the process terminates. We need to implement error recovery. Something like pre-allocating the file and writing the incoming packets in their respective offsets. If one or more packets are lost, these are re-requested to *fill in the gaps*.

DonLakeFlyer, 2016-03-24:
Should we bail out on this download code and just use the more robust log download code instead?

DonLakeFlyer, 2016-05-13:
> Should we bail out on this download code and just use the more robust log download code instead?

@LorenzMeier Opinions here?

LorenzMeier, 2016-05-13:
Right now users report the log download to be slower and more buggy. So I'd say we enable file download only under an advanced checkbox (its too handy to toss is completely) and focus on fixing log download.

DonLakeFlyer, 2016-06-14:
I just succesfully downloaded a 40M file over usb succesfully. So this is just flaky. It's going to take some work to harden. Moving to 3.1. For now users can retry, or use the slower log download.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier