mavlink/qgroundcontrol #2433: Do not connect via Radio if Autopilot is found on USB

LorenzMeier, 2015-12-20:
@DonLakeFlyer QGC currently connects via radio and USB if you connect them both (which is kind of common during setup). While in theory we want to support that on the autopilot side during operation, the setup / calibration code is not designed for this and all hell breaks loose.

The auto-connect code should only talk to the autopilot if it finds one connected directly via USB.

DonLakeFlyer, 2015-12-20:
This is not going to be easy to do. You don't know whether a link is to the same vehicle until you start talking to it and get a vehicle id back. You can't really do it from the auto-connect side of things. A way to do it would be if a vehicle has multiple links associated with it and one of those links is USB then when QGC sends mavlink messages it only sends to the USB link instead of all links. Would that work? Would it break some other use case? Is USB + Radio the only case of this problem?

LorenzMeier, 2015-12-20:
Don't connect to anything else if you find an autopilot connected. Its not normal that it is connected, so the user intended to do something specific with it.

DonLakeFlyer, 2015-12-20:
Hmm, not super understandable special case for multi-vehicle but will work and simpler change.

DonLakeFlyer, 2015-12-20:
This isn't going to work. If you have a radio hooked to your computer it will have already auto-connected to it. It's just sitting there waiting for a vehicle to show up. It's already too late to not connect to it. Radio will always be first, then USB.

DonLakeFlyer, 2015-12-20:
Back to not talking to additional links if USB is connected as possibly only option.

LorenzMeier, 2015-12-20:
Yes, and that seems to make a lot of sense as user model. By plugging the autopilot in the user told you he wants to talk to it through a fast link, and not through multiple.

DonLakeFlyer, 2015-12-21:
I'll get this done tomorrow.

DonLakeFlyer, 2015-12-28:
Ok. I'm still lost with repect to understand how multiple links to the same vehicle are supposed to work correctly. Some of the mavlink messages create a protocol of communication with the vehicle: Parameter, Mission, FTP, Log download, others? This is a back and forth communication where in some cases sequencing is important and whether a message is received or not moves that state of the protocol forward. If things are received out of sequence or multiple times it may indicate packet loss. In the case of multiple links I'm sure these little mini-protocols will work. The fact that you will get responses to the protocol on multiple links will confuse the mini-protocol state machine. In some cases you may be able to harden the protocol state handler against this. But I'm not sure it's possible in all cases to tell the difference between packet loss and multiple links. It would require a review of each mini-protocol spec and implementation. Even if you can support it, writing mini-protocol handlers that are hardened to this case will be a huge pain in the ass with way more complexity.

USB is just a special case we are trying to handle in a different way (so far unsuccessfully). Given the fact that multiple links are a supposed to be a supported thing, special casing USB (if possible) doesn't seem like the right fix. We need to make it work for real all the time for all cases. At this point I'm not sure that is possible.

So I come back to the issue of whether multiple ground station mavlink's to the same vehicle is a real use case we need to support. If so, there is a pile of work that needs to happen. I can pretty much guarantee QGC Mission and FTP mini-protocol handlers are currently broken to this case. Log Download will likely be broken unless we specifically call it out that it needs to undestand that this can happen as the new code is written. I think Parameter is ok, but needs review. Anything setup protocol (calibration and so forth) related is also broken likely both on the QGC and Firmware side. I'm not sure how to fix that. Not sure if I missed other cases.

LorenzMeier, 2015-12-29:
The real user scenario with "multiple links" is this:

  * Receive on N links
  * Send to "best" link which hasn't timed out

So from an implementation perspective this is how we could make it work:

 * Receive on all links (easy)
 * Attach a priority to each link and send only to the link with the highest priority. If the link with the highest priority changes (because USB connected or the radio connection dropped), reset all mini-protocol state machines

That way you only have one session through one link. But the vehicle will still opportunistically use the "best" link it has. In Wifi range it will use Wifi and out of Wifi range it will fall back to radio.

DonLakeFlyer, 2015-12-29:
Ok, I'll have to think about how to switch state machines on the fly. That's going to be a bit tricky. I'm not sure a reset will be possible. But keeping a state machine on the same link may be. Since they are short lived then next one will pick up the right link.

DonLakeFlyer, 2015-12-30:
I think the way to do this is for mini-protocol's to essentially keep a session id sort of a thing. The session id tracks which link to use in a multi-link situation. When a new protocol starts, it always uses the best one. But while the mini-protocol is in play i stays on the same session/link. If the session/link it's currently working on goes away, retry should kick in and the mini-protocol will move to a new link. Adds a bit more complexity but I think it shouldn't be too hard to implement.

LorenzMeier, 2016-01-03:
I really, really think we need this for the release. It will give a lot of users a very flaky experience on desktop if we don't have this in. Nobody disconnects the radio when he has the autopilot connected and a lot of people use USB to power the autopilot during testing / pre-flight / setup.

DonLakeFlyer, 2016-01-03:
I know I had it marked for Release 3.0 already.

closed 2016-01-13 by DonLakeFlyer