mavlink/qgroundcontrol #2435: Implement log download for APM Stack

DonLakeFlyer, 2015-12-21:


DonLakeFlyer, 2015-12-21:
Implement similar APMP/MP where you can see the list of available logs. Plumb this through FirmwarePlugin support so we can use the same UI for APM/PX4.

DonLakeFlyer, 2015-12-21:
Should get PX4 Stack to implement LOG_* mavlink message support as well. Then log listing and download could be mavlink generic.

LorenzMeier, 2015-12-21:
I think this warrants a call. I'm open to adding support, but its more about short-term hacks and long-term architecture and not so much about generic vs. non generic.

dogmaphobic, 2016-01-03:
I'm taking over this as having done the firmware side, I'm probably the one in best position to test as I go (I can fix both sides at once if needed). With that said, the first release will be just a dialog that allows you to pick the logs to download to your computer. Once we have a proper *Analyze" view, it could be integrated within so logs can be loaded directly into the view.

LorenzMeier, 2016-01-03:
I think just getting the file is more than enough for 99% of users. We have great GUI plotting tools already and I'm not sure this is truly GCS functionality.

closed 2016-01-17 by DonLakeFlyer