mavlink/qgroundcontrol #2450: QGroundControl 2.8 waypoint settings icon disappears

suraqis, 2015-12-24:
In flight plan mode waypoint settings icon (gearwheel) disappears when changing waypoint coordinate frame or checking out "autocontinue" checkbox.

DonLakeFlyer, 2015-12-24:
This is correct behavior. There are two modes, "friendly edit" and "raw edit". Clicking the gear takes you to raw edit where you have access to all settings. If you change a setting which can not be represented by friendly edit the gear will go away and you can't go back.

closed 2015-12-24 by DonLakeFlyer

suraqis, 2015-12-24:
tricky as for me, but ok.

DonLakeFlyer, 2015-12-24:
For usability there should probably be some way to get back to friendly edit or some indiacator as to what happened. I'll enter a separate issue for that.

suraqis, 2015-12-24:
I think that hiding icon is not the best way to show changes. Maybe it's better to deactivate and dim it.

DonLakeFlyer, 2015-12-24:
Issue #2453