mavlink/qgroundcontrol #2452: QGroundControl 2.8 fullscreen mode menu bug

suraqis, 2015-12-24:
In QGroundControl 2.8 on Windows when switched to fullscreen application menus and other drop down menus are not dropping down. When out from fullscreen menu restores.

DonLakeFlyer, 2015-12-24:
@dogmaphobic This seems to ring a bell with someone you hit way back when. Is that correct?

dogmaphobic, 2015-12-25:
Yeah, but that was iOS. I'll take a look at it.

DonLakeFlyer, 2015-12-25:
I'm building it right now. I'm actually thinking of just tossing that fullscreen thing. Let the OS handle full screen if it supports it instead of GGC.

DonLakeFlyer, 2015-12-25:
Actually it looks like Windows QtCreator on my laptop is hosed. So won't get to this till tomorrow.

dogmaphobic, 2015-12-25:
Yep, I was able to reproduce it. It appears to be a Qt bug.
https://forum.qt.io/topic/56234/combobox-and-menu-doesn-t-work-in-full-screen-qml

DonLakeFlyer, 2015-12-25:
I'd rip out fullscreen then.

dogmaphobic, 2015-12-25:
On it...

closed 2015-12-25 by DonLakeFlyer