mavlink/qgroundcontrol #2456: Validate compass offsets for correctness

DonLakeFlyer, 2015-12-25:
In Copter-3.2.1 and later offsets are considered acceptable provided their combined “length” is less than 600 (i.e. sqrt(offset_x^2+offset_y^2+offset_Z^2) < 600)

DonLakeFlyer, 2015-12-25:
@LorenzMeier Any usefullness to do the same similar thing for PX4 Compass Cal on QGC side? To idenitify bad cal, or bad compass positioning? Not sure if firmware side already does such a thing.

LorenzMeier, 2015-12-25:
Probably. The APM devs are always confused by assuming a zero offset is a good thing (which is meaningless). However, the measurement range is 1.4 gauss and the earth field strength is 0.6 gauss, so 0.6 + 0.6 = 1.2 is just below the maximum range the default sensor can handle.

DonLakeFlyer, 2015-12-25:
So what you are saying is that this is an upper bound, but in fact there is a possible lower bound as well?

LorenzMeier, 2015-12-25:
Not really, since length has no sign. So just checking for 0.6 gauss does the trick.

DonLakeFlyer, 2015-12-26:
I don't quite understand all this stuff. So just to be clear. Is it ok to add the same:

sqrt(offset_x^2+offset_y^2+offset_Z^2) < 600

check to PX4 compass cal and warn if >= 600?

LorenzMeier, 2015-12-27:
PX4 uses SI units, so it would be 0.6 gauss. I can handle that if you like - I would be more than happy to trade the implementation time for the tuning sliders.

Done: https://github.com/PX4/Firmware/commit/3bb733018beb033ff067861005ee8c5bcced34a0

If you are happy with the APM state you can close this one.

DonLakeFlyer, 2015-12-27:
This hasn't been implemented yet in APM Compass Cal.

closed 2015-12-27 by DonLakeFlyer

DonLakeFlyer, 2015-12-28:
Implemented on APM side now