mavlink/qgroundcontrol #2469: Scrollable regions need scrollbar on desktop

DonLakeFlyer, 2015-12-26:
Right now we use Flickable in a lot places. Flickable has no scrollbar. The issue with no scrollbar is that one of the main ways you are left to scroll with on desktop is mouse-wheel. If all you have is mouse-wheel it's too easy to hit a combo box when you are scrolling an then all of a sudden you are changing combo box selections when all you wanted to do was scroll. I think for desktop we need scroll bars as a mechanism to use that don't have this problem. On tablet we should not have scroll bars.

dogmaphobic, 2015-12-26:
You can click and drag to move a Fickable, in the same way you touch and drag it on mobile platforms. With that said, I agree that this behavior is not *natural* for the desktop.

DonLakeFlyer, 2015-12-26:
Yeah I know. It's a crappy experience though. Trying to scroll airframe config in a reasonable way is next to impossible. Not really sure scroll bars will fix the probliem. Airframe config for example may just be a bad design with too many combos on a big scrollable reason.

closed 2016-01-17 by DonLakeFlyer