mavlink/qgroundcontrol #2470: can't connect to PixRacer over USB on Ubuntu 15.10

kd0aij, 2015-12-26:
With latest daily build: 2.8.0-318-g9073ac2:

can connect to /dev/ttyACM0 with a serial terminal, so USB connection is good.
manually executing "sh /etc/init.d/rc.usb" on ACM0 starts mavlink; 
disconnecting serial terminal and starting QGC results in autoconnect over USB, but no parameters are received. QGC complains "Vehicle did not respond... retrying" but never succeeds

LorenzMeier, 2015-12-26:
Did you uninstall the modem manager?

kd0aij, 2015-12-26:
I had double checked that, triple checking:
```
sudo apt-get remove modemmanager
Package 'modemmanager' is not installed, so not removed
```
Hope I didn't misspell it

DonLakeFlyer, 2015-12-26:
It's because PixRacer is a new board type which isn't fully plumbed through QGC. The rc.usb script is not pushed out by QGC for all boards. Just identified pixhawk boards. There is a missing case in:

bool QGCSerialPortInfo::boardTypePixhawk(void) const

for V4 boards.

DonLakeFlyer, 2015-12-26:
Should be fixed now.

closed 2015-12-26 by DonLakeFlyer

kd0aij, 2015-12-27:
no luck with 2.8.0-320 either
Also having trouble connecting a serial terminal to ttyACM0 now too. Must be something wrong on the pixracer end. Autoconnects fine to an fmuv2 on the same USB port and cable.

kd0aij, 2015-12-27:
running rc.usb on the nsh console works to establish a connection to mavproxy on fmuv2:
```
mavlink start -r 800000 -d /dev/ttyACM0 -m config -x
mavlink_if1: mode: 5, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
```
But shouldn't the baud rate be 921600?

Doing the same thing on PixRacer results in "waiting for heartbeat" from mavproxy, so it seems that the PixRacer isn't sending a heartbeat