mavlink/qgroundcontrol #2482: Boot QGC, Link Configuration "LinkConfigurations/Link0" has no type

DonLakeFlyer, 2015-12-28:
This will happen even if you do a --clear-settings. Also these are qWarning's which I don't think is correct usage here. qWarning is more like QGC is about to die. Whereas I think these checks are more like bad settings. Not quite sure.

dogmaphobic, 2015-12-28:
I don't see that message. I just merged master and it all seems normal. That warning can only be shown if you have a corrupt ini file. If you delete the ini file (*--clear-settings*), I can't see how that loop would even run (there should be nothing to read).

This is from the PixRacer (it shows both the USB and UDP heartbeats and the proper USB only link). No warnings. What are the steps to reproduce it?

```
Starting /Volumes/RAMDisk/build-qgroundcontrol-Desktop_Qt_5_5_1_clang_64bit-Debug/debug/qgroundcontrol.app/Contents/MacOS/qgroundcontrol...
Logging ini file directory "/Users/gus/Library/Preferences/QtProject"
Filter rules "qt.network.ssl.warning=false\n"
Settings location "/Users/gus/Library/Preferences/com.org-qgroundcontrol.QGroundControl.plist" true
Mapping cache directory: "/Users/gus/Library/Caches/QGCMapCache55"
0
Orphaned log file count 0
LinkManagerLog: New auto-connect UDP port added
LinkManagerLog: Waiting for next autoconnect pass "/dev/cu.usbmodem1"
Temp log "/var/folders/k6/cxtnktrn32sf18mdxtm13msc0000gn/T/FlightData793802.mavlink" false
LinkManagerLog: New heartbeat on link:vehicleId:vehicleMavlinkVersion:vehicleFirmwareType:vehicleType  "Default UDP Link" 1 3 12 1
LinkManagerLog: New auto-connect port added:  "Pixhawk on cu.usbmodem1" "/dev/cu.usbmodem1"
LinkManagerLog: New heartbeat on link:vehicleId:vehicleMavlinkVersion:vehicleFirmwareType:vehicleType  "/dev/cu.usbmodem1" 1 3 12 1
```

DonLakeFlyer, 2015-12-28:
> --clear-settings), I can't see how that loop would even run

That's what I thought as well. Let me try again, now that you say that I may have had a typo in my command line.

DonLakeFlyer, 2015-12-28:
Ok. I must have had a typo in my command line. Something is causing this to happen though. Not sure what. I was in this state of both my main machine and my laptop. Closing for now to see if it comes back.

closed 2015-12-28 by DonLakeFlyer