mavlink/qgroundcontrol #2491: Airframe configuration pictures - OS X QGC 2.8.0-346-g9ae456c (Development) build

lvale, 2015-12-29:
Need more variants and pics for airframe selection (copter) for APM

Current:

[image]

Mission Planner 

[image]

APM Planner

[image]

lvale, 2015-12-29:
Linked to #2440

DonLakeFlyer, 2015-12-29:
The other issue here is that the frame types show up differently based on what firmware you are running.

lvale, 2015-12-29:
I'm running Hexa so perhaps Hexa frames should come out ?

DonLakeFlyer, 2016-01-02:
I"ve dumbed down the UI for now with no images. Coming at some point will be auto-generate images which include correct prop rotation, motor numbers and pixhawk wiring. Once we have that this will get nicer again.

closed 2016-01-02 by DonLakeFlyer