mavlink/qgroundcontrol #2512: Support for new VTOL mission commands and flight modes

sanderux, 2016-01-02:
For both PX4 and APM the following commands are/will be available soon:

               <entry value="84" name="MAV_CMD_NAV_VTOL_TAKEOFF">
                    <description>Takeoff from ground using VTOL mode</description>
                    <param index="1">Empty</param>
                    <param index="2">Empty</param>
                    <param index="3">Empty</param>
                    <param index="4">Yaw angle in degrees</param>
                    <param index="5">Latitude</param>
                    <param index="6">Longitude</param>
                    <param index="7">Altitude</param>
               </entry>
               <entry value="85" name="MAV_CMD_NAV_VTOL_LAND">
                    <description>Land using VTOL mode</description>
                    <param index="1">Empty</param>
                    <param index="2">Empty</param>
                    <param index="3">Empty</param>
                    <param index="4">Yaw angle in degrees</param>
                    <param index="5">Latitude</param>
                    <param index="6">Longitude</param>
                    <param index="7">Altitude</param>
               </entry>

For APM the following flight modes will be added soon:

mode 17: QSTABILIZE (like copter stabilize)
mode 18: QHOVER (like copter alt_hold)
mode 19: QLOITER (like copter loiter)

DonLakeFlyer, 2016-01-02:
Thanks. Lorenz just added transition here: https://github.com/mavlink/qgroundcontrol/pull/2510

LorenzMeier, 2016-01-03:
They are here: https://github.com/mavlink/mavlink/commit/c81d2fd794d1a304a85a9e6769892a37bef71c1a