mavlink/qgroundcontrol #2514: Provide autopilot ID and GCS ID and version on Firmware download

LorenzMeier, 2016-01-03:
@DonLakeFlyer For the sake of Dronecode statistics it would be great to extend the URL with get params like:

http://binary.px4?ap=<AUTOPILOT TYPE STRING>&id=<AUTOPILOTUID>&stack=<FIRMWARE STACK ID>

For QGC statistics themselves, it would be great if it would open the QGC website on boot to check for updates like:

http://qgroundcontrol.org/updatecheck/?id=<AUTOPILOTUID (= the first it has flashed)>

LorenzMeier, 2016-01-03:
I will likely use this to add tracking. https://github.com/binaryage/leechgate

LorenzMeier, 2016-01-04:
Discussing the MP format and potential adjustments with Michael here:
https://github.com/diydrones/MissionPlanner/issues/1093

Goal is to have a simple, well-documented API how to pull firmware files so Dronecode gets a good picture of where things are going.

DonLakeFlyer, 2016-01-04:
"Only Android? Why not everything? I have all the code for it. I can do it in a day or so for all platforms at once. Anonymous, non identifying usage analytics."

@dogmaphobic Just stumbled across the Google thing and wanted to make a note of it. Do you have a server side backend though for analysis? How does that work?

dogmaphobic, 2016-01-04:
It's all hosted by google. You setup an *analytics* account, which will give you a *key* to use. It's a bit of a daunting learning curve but it's somewhat logical. From the app, you send http requests to a specific (google) API. It includes, along with your *key*, *events*, *categories*, etc. From your *Analytics Portal*, you can monitor usage of various aspects of the app (based on those *events*, *categories*, etc.)

I assume @LorenzMeier would create the *key*. This would only go into the release build. Non release builds would have a blank key and would not generate any analytics calls.

https://developers.google.com/analytics/?hl=en

DonLakeFlyer, 2016-01-21:
Narrowing to Firmware Usage statistics for 3.0 release. Re-opened #2521 as issue for UI tracking.

DonLakeFlyer, 2016-01-21:
You talk about checking for updates. Checking for updated what? QGC doesn't do any update checking on boot.

DonLakeFlyer, 2016-03-25:
http://s3bucket/binary.px4?ap=<AUTOPILOT TYPE STRING>&id=<AUTOPILOTUID>&stack=<FIRMWARE STACK ID>

DonLakeFlyer, 2016-03-25:
@LorenzMeier I assume you are talking about firmware download stats using this:
```http://s3bucket/binary.px4?ap=<AUTOPILOT TYPE STRING>&id=<AUTOPILOTUID>&stack=<FIRMWARE STACK ID>```

But for firmware download, the extra param would be ground station name wouldn't it? The url to the firmware itself already dictates thing like autopilot type. Also a firmware download time the autopilotuid isn't known? Since all  you have is the bootloader.

dagar, 2016-07-12:
S3 now has logging enabled for both the px4-travis and qgroundcontrol buckets. We'll likely need some 3rd party tool to analyze, but I think it should have most of information you want.

closed 2026-10-04 by github-actions[bot]