mavlink/qgroundcontrol #2515: QGC crash when one clicks on Radio button when connected to SITL

pritamghanghas, 2016-01-03:
Does the same thing with raspilot board running APM. I couldn't check with pixhawk. Don't have a working pixhawk right now.
 
Here is the stack trace as shown by QtCreator

0	QVariant::QVariant(QVariant const&)			0x7ffff32cbd90	
1	Fact::rawValue	Fact.h	81	0x53e517	
2	RadioComponentController::_setInternalCalibrationValuesFromParameters	RadioComponentController.cc	751	0x6c7d9d	
3	RadioComponentController::_stopCalibration	RadioComponentController.cc	930	0x6c932d	
4	RadioComponentController::start	RadioComponentController.cc	129	0x6c558c	
5	RadioComponentController::qt_static_metacall	moc_RadioComponentController.cpp	287	0x7bbd35	
6	RadioComponentController::qt_metacall	moc_RadioComponentController.cpp	506	0x7bc6af	
7	??			0x7ffff416d78c	
8	??			0x7ffff40f1cfd	
9	??			0x7ffff40f2dd2	
10	QV4::QObjectMethod::callInternal(QV4::CallData *)			0x7ffff40f3977	
11	QV4::Runtime::callProperty(QV4::ExecutionEngine *, int, QV4::CallData *)			0x7ffff41079d5	
12	??			0x7ffff40f85fd	
13	??			0x7ffff40f92d1	
14	??			0x7ffff40a804f	
15	??			0x7ffff41945fe	
16	QQmlBoundSignalExpression::evaluate(void * *)			0x7ffff413dcde	
17	??			0x7ffff413e353	
18	QQmlNotifier::emitNotify(QQmlNotifierEndpoint *, void * *)			0x7ffff4178ab7	
19	QQmlData::signalEmitted(QAbstractDeclarativeData *, QObject *, int, void * *)			0x7ffff411ebbc	
...	<More>

pritamghanghas, 2016-01-03:
ok found the problem. function RadioComponentController::_chanMax(void) return 14 as the number of channels and then it tried to get RC9_TRIM. Such parameter don't exist beyond RC8. So it returns null and then null ptr referencing and crash.

pritamghanghas, 2016-01-03:
This is trange. RC9* parameters don't exist. While RC10 and RC11 exist and after that nothing. Would it make sense to reduce _chanMax to 8 for APM. Though pointer checking may still be required for missing parameters. If a parameter is missing for RC#, What do we do that make the fields to some  default or 0 in channelInfo struct.

DonLakeFlyer, 2016-01-03:
QGC will only work with the latest ArduCopter or above. I think that's 3.3. Anything below that is going to be missing parameters. I haven't been trying to support versions below that because it just gets to complicated dealing with the differences.

http://copter.ardupilot.com/wiki/arducopter-parameters/

DonLakeFlyer, 2016-01-03:
There needs to be a version check put in for ArduPilot versions. If the version is too old it should fall back to generic mavlink support.

pritamghanghas, 2016-01-03:
On ArduCopter master. this http://copter.ardupilot.com/wiki/arducopter-parameters/. This doesn't contain CH9_MIN CH9_MAX or CH9_TRIM
some how ch10* and ch11* are there. Plus after that there is nothing. So no way 14.

DonLakeFlyer, 2016-01-03:
Need to look at the RadioComponent code since it should trigger a "missing params" alert instead of crashing.

pritamghanghas, 2016-01-03:
trigger missing parameter message but after that returns null for the FACT
which gets deferenced.

DonLakeFlyer, 2016-01-03:
It's not doing the check ahead of time like it should.

pritamghanghas, 2016-01-03:
I can do that, But I still don't get the number 14 for channel count and
how its supposed to be used, when there is currently no way to use them or
store any information about them.

DonLakeFlyer, 2016-01-03:
It's going to be tricky to fix this. Since it seems like the problem is not only in master but in 3.3 as well. Both the RadioComponent and the unit test will need to be updated. Since the code is shared between PX4 and APM it's going to be a real pain. I"ll figure out how to deal with it.

DonLakeFlyer, 2016-01-06:
Flight Mode config is also referencing these missing params.

DonLakeFlyer, 2016-01-10:
@pritamghanghas Can you tell me again what params are missing when you go to Flight Mode config on your linux board? I'm not seeing references to the same missing params in the code.

DonLakeFlyer, 2016-01-10:
@pritamghanghas Once you have that can you enter a new Issue with that in it? I'm closing this one since the Radio Config is fixed.

closed 2016-01-10 by DonLakeFlyer