mavlink/qgroundcontrol #2518: QmlObjectListModel needs memory management.

dogmaphobic, 2016-01-04:
Putting here so I won't forget. Items in QmlObjectListModel are not cleaned up, causing memory leaks.

DonLakeFlyer, 2016-01-05:
This is correct behavior. The caller is respondible for memory management when using QmlObjectListMode.

dogmaphobic, 2016-02-19:
Looking further into this, the issue is with methods like `clear()` and `deleteRows()`. They remove elements without deleting them. The problem is figuring out if it's safe to delete (same thread) or not (having to call `deleteLater()`). I guess we could do that by comparing threads:

Instead of:

```cpp
void QmlObjectListModel::clear(void)
{
    while (rowCount()) {
        removeAt(0);
    }
}
```

Do something like this:

```cpp
void QmlObjectListModel::clear(void)
{
    while (rowCount()) {
        QObject* removedObject = removeAt(0);
        if(removedObject->thread == this->thread())
            delete removedObject;
        else
            removedObject->deleteLater();
    }
}
```

NaterGator, 2016-02-19:
Is there a particular reason to care? Just call deleteLater every time and don't worry about which thread you are on.

In fact, is there even a threading concern? Why would this ever be used on a non GUI thread?

dogmaphobic, 2016-02-19:
Map tile sets are created on a different thread and are inserted into a QmlObjectListModel. In that particular case, I `moveToThread()` before sending them down (from the tile worker thread to the tile manager) but it can't be guaranteed all future uses will do the same.

As to always calling `deleteLater()`, good question. I do not know the behavior of that call when called from the same thread.

NaterGator, 2016-02-19:
Calling deleteLater on the main thread is perfectly safe provided you can tolerate two conditions:
1) The object will not actually be deleted until control returns to the event loop handler
2) The event loop must be running or the controlling thread must call sendPostedEvents periodically.

On the GUI thread we damn well better be running the event loop and given we're talking about QML I think we can tolerate the C++ objects living for a few extra scopes.

As a matter of safety I almost always invoke deleteLater when destroying a QObject manually.

dogmaphobic, 2016-02-19:
That's what I thought but I was too ~~lazy~~ busy to check Qt's source. Assuming @DonLakeFlyer is fine with this, I can do these changes.

DonLakeFlyer, 2016-02-19:
Give me some time to look at the code that uses this. There was some reason I didn't implement memory management in it. I just can't quite remember what the hell that was at the moment.

DonLakeFlyer, 2016-02-22:
I believe to not do it was to work around the C++->Qml object lifetime issues causing binding error messages on the Qml side. Using deleteLater doesn't fix that problem because the deleteLaters happen before lazy Qml destruction happens. That said I still think this may be doable, but it going to require some a serious look through and testing all the mission editing code with such a thing to see what happens. So let me take this, and I'll take a look during 3.1 release.

NaterGator, 2016-02-22:
@DonLakeFlyer it looks like this model is not adhering to the required signaling specification. The clear method is invalidating data in the model without informing anything attached that the underlying data has changed. That is likely why QML is not properly managing ownership/lifecycle. According to the [QQmlEngine](http://doc.qt.io/qt-5/qqmlengine.html#ObjectOwnership-enum) docs ownership should automatically be assigned to C++. 

I strongly suspect if we fix the QmlObjectListModel to emit the requisite signals before / after manipulating the underlying data this issue will be resolved.

DonLakeFlyer, 2016-02-22:
It's not an ownership problem. The issue is lazy destruction of Qml. The sequence is as follows:
1) Load Qml component
2) Qml component references an object on the C++ side
3) Unload Qml component
4) Delete C++ object

This will cause a binding error. The reason is that step 3 does not actually destroy the Qml. It's still hanging around with bindings still connected up to everything. It will eventually be lazy destructed but it takes two passes through an idle event loop for that to happen. Hence the C++ side has a really hard time knowing when it can destroy stuff.

DonLakeFlyer, 2016-02-22:
> The clear method is invalidating data in the model without informing anything attached that the underlying data has changed.

That's because QmlObjectListModel doesn't many it's own memory. If you call clear you need to have managed destruction yourself somehow after the fact. As far as I know the abstract list mode implementation in QmlObjectListModel isn't missing any signalling.

DonLakeFlyer, 2016-02-22:
That C++/Qml lifetime problem I describe above is why at shutdown you get binding errors. I've handled cases such that you don't get them during normal running. By jumping through crazy hoops on Vehicle change/delete for example. But it wasn't worth the effort to deal with it at shutdown.

DonLakeFlyer, 2016-09-13:
@NaterGator Looking at this again. In relation to the clear method not signalling properly. It seems to be doing the right thing:
```
void QmlObjectListModel::clear(void)
{
    while (rowCount()) {
        removeAt(0);
    }
}
```
Which will then cause all the right things to happen if you follow it through. I don't see where this is wrong?

DonLakeFlyer, 2017-04-06:
Given all the code in there that assumes this is not automatic as well as the fact that there are many cases where that is not really what you want this isn't going to happen.

closed 2017-04-06 by DonLakeFlyer