mavlink/qgroundcontrol #2520: Need some mechanism to capture logging from Android builds

DonLakeFlyer, 2016-01-04:


DonLakeFlyer, 2016-01-04:
Will consider for 3.0, but I think it's unlikely.

dagar, 2016-01-04:
Which logging are you referring to?

We do get some crash information if the user chooses to submit.

https://play.google.com/apps/publish/?dev_acc=03583851360351630691#ErrorClusterDetailsPlace:p=org.mavlink.qgroundcontrol&et=CRASH&lr=LAST_6_MONTHS&ecn=Native+crash&tf=unknown&tc=unknown&tm=unknown&nid=57f3a59d758e6758&an&c&s=new_status_desc

DonLakeFlyer, 2016-01-04:
QLoggingCategory logging.

dogmaphobic, 2016-01-04:
It goes out to logcat. At runtime, you can run ```adb logcat``` and all logging (including QGC's) will be output. You can even capture it using WiFi if you don't have the device connected directly (USB).

logcat: http://developer.android.com/tools/help/logcat.html
adb: http://developer.android.com/tools/help/adb.html

DonLakeFlyer, 2016-01-04:
> It goes out to logcat.

Not helpful on users machine. They won't have that installed. I'm talking about using logging to debug a problem a user is having.

DonLakeFlyer, 2016-01-04:
I'd guess the logcat logs are sitting on the tablet somewhere. Wonder if there is a tablet app that you can run to look at them?

DonLakeFlyer, 2016-01-21:
I'm planning to move this to 3.1 release. Speak up now if you disagree.

DonLakeFlyer, 2016-04-09:
Build on Nate's stuff.

DonLakeFlyer, 2016-05-13:
I think this is going to be needed for 3.0 in order to aid debuting android user problems

closed 2016-05-25 by DonLakeFlyer