mavlink/qgroundcontrol #2524: Don't start another QGC instance if the UDP default port is taken

LorenzMeier, 2016-01-04:
@DonLakeFlyer @dogmaphobic I have used this before, it fixes multiple instances on all OS and works perfectly.

DonLakeFlyer, 2016-01-10:
This will only sort of work since you can turn off UDP auto-connect. @dogmaphobic Did you say you had some way to do this?

dogmaphobic, 2016-01-13:
The problem is that you can configure a different port. If you do that, there is no simple way to find it (the config is only loaded way later in the game and this test has to be done very early.

I added this code to main() and it works but it implies assigning another port just for this (14551 in this case):

```
#ifndef __mobile__
    //-- Test for another instance already running. If that's the case, we simply exit.
    QHostAddress host("127.0.0.1");
    QUdpSocket socket;
    if(!socket.bind(host, SINGLE_INSTANCE_PORT, QAbstractSocket::DontShareAddress)) {
        qWarning() << "Another instance already running.";
        exit(-1);
    }
#endif
```

If that's acceptable, I have it ready for a PR.

dogmaphobic, 2016-01-13:
On Mac OS, the code above even brings the existing instance to the foreground.

DonLakeFlyer, 2016-01-13:
I don't know much about whether taking another port is ok. @LorenzMeier ?

DonLakeFlyer, 2016-01-21:
@LorenzMeier ?

dogmaphobic, 2016-01-23:
Done

closed 2016-01-23 by dogmaphobic

paan2097, 2017-08-31:
I need to run multiple instances of QGroundControl or multiple HIL configs. I have two separate instances of XPlane up and running, but now need to use either two HIL configs or separate QGroundControls to run multple HITL simulations on a single computer. Can anyone advice on how I can get this working? Thanks.