mavlink/qgroundcontrol #2525: Mode switching doesn't work in HITL, shows multiple items ticked if it fails

LorenzMeier, 2016-01-04:
I'm not clear yet where things break, but once PX4 is in HITL mode mode switches are completely ignored. Might be a firmware thing, the reason I'm filing it here is a different one: The mode switch indicator still updates, resulting in two modes being selected even if the system rejected or didn't respond.

DonLakeFlyer, 2016-01-05:
The fact that the mode switch indicator update is just a bug. That dropdown cna't have checkmarks on it. I have a fix for this. I stepped through the actual flight mode change onthe QGC side and everythign looks good from there. The change of mode is going out. The firmware just never does anything with it. That's going to need to be debugged fromthe firmware side. Also when I ran HIL with XPlane firmware spits out tons of ekf messages.

DonLakeFlyer, 2016-01-05:
There is also a crashing bug in QGC HIL when you shutdown. I'll fix that as well.

DonLakeFlyer, 2016-01-05:
FYI: I did get it to switch to Loiter one time. No idea what I did to make that happen. It only worked once.

DonLakeFlyer, 2016-01-05:
I'm trying to debug the firmware side.

DonLakeFlyer, 2016-01-05:
Here is what I found:
- If you attempt to change flight mode before you connect HIL (which turns on HIL mode) the flight mode changes are rejected by firmware. So the messages are getting to the board.
- If you attempt to changle flight mode after HIL is turned on firmware no longer responds to messages from QGC. Not on does flight mode change no longer work, but things like param refresh are not responded to either. I validated that in this case the flight mode change message is going out from the QGC side. No idea what is going on here.

LorenzMeier, 2016-01-05:
Thanks, that cross validation is helpful. Its some sort of regression. I need to instrument the receive logic if we suffer from a buffer overflow or if the vehicle's HIL state machine somehow disables command handling in a bad way.

DonLakeFlyer, 2016-01-05:
> I need to instrument the receive logic if

I intrumented commander and it didn't seem like the flight mode message ever got  there. After that I gave up.

closed 2016-01-05 by DonLakeFlyer