mavlink/qgroundcontrol #2526: Integrate survey code

DonLakeFlyer, 2016-01-04:
From David Goodman:

I've released the code available here: http://dagoodma.github.io/dppl_code/. The QGC branch still needs to be merged and fixed up a little before a pull request, but it's available here: https://github.com/dagoodma/qgroundcontrol/tree/dpp_addon_dtspascpp.

dagar, 2016-01-04:
I don't like the idea of pulling dppl in as a binary and passing around json. I'll try to get it building with qmake and then include it as a submodule.

DonLakeFlyer, 2016-01-04:
It should just be in the main tree. Shouldn't be a submodule either. I plan to look at the code and re-work as needed to bring it in. I'll handle bringing this in after the 3.0 release goes out. Don't want it before then.

DonLakeFlyer, 2016-01-04:
@dagar I'll ping you when I get to this for help.

DonLakeFlyer, 2016-01-05:
Related to Issue #2507

DonLakeFlyer, 2016-02-27:
Tracking in #2507

closed 2016-02-27 by DonLakeFlyer