mavlink/qgroundcontrol #2532: Settings are being stored in different (and some wrong) places.

dogmaphobic, 2016-01-04:
Adding here so I won't forget.

The settings file reported by QSettings has a weird filename:

```
Settings location "/Users/<me>/Library/Preferences/com.org-qgroundcontrol.QGroundControl.plist"
```

Note the _**com.org**_ prefix, which should be the *domain* TLD.

In addition, two separate files are being written to ```/Users/<me>/Library/Preferences```. The one above and another named ```org.qgroundcontrol.qgroundcontrol.plist``` (with a proper TLD).

With that said, there is some other settings file somewhere as the screen position, audio mute, etc. are only reset when you select *Clear all settings on next start* from the General Settings panel.

Just opening and closing QGC generates over 1,700 file opens. Of notice (I didn't spend too much time digging):

```
/Users/<me>/Library/Saved Application State/org.qgroundcontrol.qgroundcontrol.savedState/windows.plist 
/Users/<me>/Library/Saved Application State/org.qgroundcontrol.qgroundcontrol.savedState/data.data 
/Users/<me>/Library/Saved Application State/org.qgroundcontrol.qgroundcontrol.savedState/restorecount.plist 
/Users/<me>/Library/Autosave Information/org.qgroundcontrol.qgroundcontrol.plist 
```

DonLakeFlyer, 2016-01-13:
The change to move the settings location on OSX is screwin up a bunch of other stuff as well. The parameter meta data is stored in the same location. By using the .ini format Qt creates a directory for all this stuff with the org naming. Then you have everything in one place. By switching to plist it no longer creates a directory and instead dump everything in Preferences which could then lead to file naming collisions. It also doesn't allow you to toss all caches settigns by just removing a directory. Now you have to poke around in individual files.

Can we switch back to the way it was before?

dogmaphobic, 2016-01-13:
>Can we switch back to the way it was before?

Done in #2600

closed 2016-01-13 by dogmaphobic