mavlink/qgroundcontrol #2538: Log download gets stuck in a loop on APM

DonLakeFlyer, 2016-01-05:
Get's a couple logs in the list then spits out of continuous stream of these:
```
Received log entry for out-of-bound index: 600
Received log entry for out-of-bound index: 601
Received log entry for out-of-bound index: 602
Received log entry for out-of-bound index: 603
Received log entry for out-of-bound index: 604
Received log entry for out-of-bound index: 605
Received log entry for out-of-bound index: 606
Received log entry for out-of-bound index: 607
Received log entry for out-of-bound index: 608
Received log entry for out-of-bound index: 609
Received log entry for out-of-bound index: 610
Received log entry for out-of-bound index: 611
Received log entry for out-of-bound index: 612
Received log entry for out-of-bound index: 613
Received log entry for out-of-bound index: 614
Received log entry for out-of-bound index: 615
Received log entry for out-of-bound index: 616
Received log entry for out-of-bound index: 617
Received log entry for out-of-bound index: 618
Received log entry for out-of-bound index: 619
Received log entry for out-of-bound index: 620
Received log entry for out-of-bound index: 621
Received log entry for out-of-bound index: 622
Received log entry for out-of-bound index: 623
Received log entry for out-of-bound index: 624
Received log entry for out-of-bound index: 625
```

dogmaphobic, 2016-01-05:
Yep, I need to flash something with APM to test this...

DonLakeFlyer, 2016-01-05:
Figured I'd give it a try to see if you got lucky

LorenzMeier, 2016-01-05:
It would be fantastic if you could put up a few words on the protocol in this repo using this fantastic markup notation:
https://github.com/mavlink/mavlink_devguide/blob/master/concept-protocol-mission.md

Which renders to this:
http://dev.mavlink.io/concept-protocol-mission.html

dogmaphobic, 2016-01-05:
> Which renders to this:

That's quite a bit of sorcery :) I had in mind to put something down (I had even asked Don where I could write it up) as the protocol is not quite obvious. I will wait I until I figure out what's wrong with APM and then write it up. While at it, how much breakage would it cause to bump the payload from 90 bytes to the maximum allowed? That is a big hit in performance.

LorenzMeier, 2016-01-05:
Why is it a hit in performance? Why can't we just put the packets back to back?

dogmaphobic, 2016-01-05:
At minimum, the header versus payload ratio is higher. For UDP, a full 1,500-byte frame is sent each time regardless.

LorenzMeier, 2016-01-05:
I am fully aware. But nothing keeps you from holding off a few more microseconds before sending off the frame. Then if the UART buffer on the AP side is large enough you should get 200-500 bytes at a time (the full buffer).

dogmaphobic, 2016-01-05:
Hmm... Something to tweak at the 8266 firmware level. I'll see what I come up with. I don't exactly have threads and interrupt routines cannot mess with (non re-entrtant) system calls. Nice puzzle though :)

dogmaphobic, 2016-01-05:
@DonLakeFlyer I just flashed APM in one of the Pixhawks. Do you have the file hierarchy (in the MicroSD card) where the log files are stored? Along with their naming. I have to create a bogus list as there is no way I can generate those *normally*.

DonLakeFlyer, 2016-01-05:
Turn off arming checks and arm the vehicle and you should get some logs.

dogmaphobic, 2016-01-16:
This has been fixed! ( #2547 )

closed 2016-01-16 by dogmaphobic