mavlink/qgroundcontrol #2542: qgroundcontrol.tar.bz2 built by travis-ci doesn't work

dagar, 2016-01-06:


dagar, 2016-01-06:
It seems to be something to do with libQt5Core.so.5, which is the only lib that's different from my local machine.

DonLakeFlyer, 2016-01-06:
@pkocmoud Was saying it only showed up recently. Maybe a week old. Something recent broke it.

pkocmoud, 2016-01-06:
Initially I thought it was a bad daily build, since the stable release works fine. I plan on installing the most modern non-LTS version of Ubuntu as that may have a better video driver. However, the LTS is the better solution in my opinion.

DonLakeFlyer, 2016-01-06:
> may have a better video driver

The video driver thing is bogus, sorry about that. Both Daniel and I can reproduce this one our machine. Daily build is just broken.

pkocmoud, 2016-01-06:
So, should I delay the OS upgrade?

DonLakeFlyer, 2016-01-06:
It won't fix the problem.

DonLakeFlyer, 2016-01-06:
I was just building my own version of Qt on Ubuntu and got this while running configure:
```
Running configuration tests...
The test for linking against libxcb and support libraries failed!
 You might need to install dependency packages, or pass -qt-xcb.
 See src/plugins/platforms/xcb/README.
```

Seems vaguely similar to the above problem?

DonLakeFlyer, 2016-01-09:
@dagar Any news on this one? Want' to announce QGC APM Beta, but need this fixed for that.

dagar, 2016-01-10:
I still don't why it's broken when created on travis-ci, but works when created locally. I'll try another source of Qt for travis.

dagar, 2016-01-11:
I believe the linux package is missing qt plugins, but finding them locally. Working on a fix.

PR #2577 (2016-01-13): include qt plugins in linux package
 - fixes #2542
commit 744b4592 (2016-01-13): include qt plugins in linux package

 - fixes #2542
closed 2016-01-13 by DonLakeFlyer