mavlink/qgroundcontrol #2548: Mission Editor: Pan/Scroll gets stuck and then you can't click on any map buttons

DonLakeFlyer, 2016-01-06:
@AndreasAntener Are you running on a Mac and panning with a trackpad?

AndreasAntener, 2016-01-06:
Yup

DonLakeFlyer, 2016-01-06:
To get it unstuck, do some small two finger zoom things a couple time and eventually it will come back. Looks like Qt bug. Going to have to figure out a workaround.

DonLakeFlyer, 2016-01-15:
This looks to be a problem when you use the Map control in our hybrid QQuickWidget setup. It doesn't happen in a native qml window. Sigh!

DonLakeFlyer, 2016-01-15:
Yup, I verified it by hacking desktop to run as Native Qml.

DonLakeFlyer, 2016-01-17:
This problem shows up on all Flickable/ScrollView/ListView containers scrolled using a Mac trackpad.

DonLakeFlyer, 2016-01-21:
The fix for this is not going to make it into 3.0 release. Needs to wait for widget conversion to Qml.

DonLakeFlyer, 2017-03-03:
Check if this is still there with 5.8

DonLakeFlyer, 2017-03-04:
Still not working correctly in 5.8. We'll live without pinch on mac trackpad

closed 2017-03-04 by DonLakeFlyer