mavlink/qgroundcontrol #2551: Limit Play store installs to 7" tablets or larger

DonLakeFlyer, 2016-01-06:
Hopefully there is some way to do this.

dagar, 2016-01-06:
Seems straightforward - 
http://developer.android.com/guide/topics/manifest/supports-screens-element.html
http://developer.android.com/guide/practices/screens-distribution.html

I did some testing with my Nexus 6p (5.7" screen) and it's nearly usable, but a few things are off screen (mission planner, etc). So for the release I'll set it to tablet only, but we should try to phase in phone support as early as possible because the Android tablet market is pretty small.
The beta test currently has ~ 50 installs and only half of them are on 7" tablets or larger.

lvale, 2016-01-09:
I tend to use more my mobile phone (OnePlus One 5.5" screen) than my Android 10" tablet, even as @dagar says it can get some elements out of place.

DonLakeFlyer, 2016-01-10:
So what's the consensus, don't look out below 7" since it "sort of" works? If so, we need a much more prevalent warning to 7" and below usage being on an as-is basis for now.

LorenzMeier, 2016-01-10:
I would be wary about running on anything that's not officially supported. Users will still complain and remember the bad experience. Either put up the work to support it or do not offer it in the first place.

DonLakeFlyer, 2016-01-10:
I'm kind of in the same camp as Lorenz. So I say we lock it down from the store. I assume if someone really really really wanted phone support (and can't tether to a larger tablet) then they could install the .apk manually.

lvale, 2016-01-11:
Are there any real usage stats available to check which %% of larger tablets are used, besides what @dagar mentioned above ?

PR #2575 (2016-01-11): Android limit to tablets for now
- fixes #2551
commit 4592b65d (2016-01-11): Android limit to tablets for now

- fixes #2551
closed 2016-01-11 by DonLakeFlyer