mavlink/qgroundcontrol #2553: manual control (joystick) forces manual mode in PX4 HIL

LorenzMeier, 2016-01-07:
@DonLakeFlyer I have labeled this all firmware, but it could be well a PX4 bug: When the joystick is enabled, the arriving manual_control packets force the autopilot into manual mode.

Is there any sort of mode setting associated with the joystick code, or does it just send manual control messages? Could this be the button to flight mode functionality going wrong?

This is the follow-up issue to the flight mode setting issue we had earlier. Here is the complete description from the user reporting it:
https://groups.google.com/d/msg/px4users/WRuXu7LPfag/Qx8flzLDEwAJ

DonLakeFlyer, 2016-01-11:
The reason HIL is not working correctly or flaky is because the rate at which QGC is sending HIL messages to the firmware seems to be choking out the connection on the firmware side. All the right messages are being sent by QGC they just are never seen by the firmware. I hacked QGC to stop sending the high rate HIL messages and then everything works. Change modes are seen, param request goes through, etc.

LorenzMeier, 2016-01-11:
Thanks! Let me see if I can fix this for good.

DonLakeFlyer, 2016-01-11:
FYI: There still could be some joystick issue here. But that main message loss problem needs to be fixed before that can really be looked at.

LorenzMeier, 2016-01-12:
We almost need a outgoing MAVLink inspector widget for this. I think that QGC is flooding the link with something other than the HIL packets, because even if I limit them to 50 Hz its still not able to keep up. Firmware does read them immediately, so I suspect there is a stream of messages I'm not expecting which is screwing us over.

DonLakeFlyer, 2016-01-12:
Just qDebug the Vehicle::_sendMessage. I only had to kill two message ids. Can't remember which ones.

DonLakeFlyer, 2016-01-21:
I'm going to give this a try again with the recent message sending changes.

DonLakeFlyer, 2016-01-21:
[image]

The receive rate for HIL spikes up to over 100hz. Doesn't that mean that the HIL setup rate for XPlane is to high in XPlane settings?

nsblackwell, 2016-04-30:
Any chance there's any news on this? If I can try killing some of the message IDs, any idea which ones?

DonLakeFlyer, 2016-07-08:
@LorenzMeier XPlane sends updates to QGC at the rate specified by the XPlane setup. QGC then turns these into mavlink messages and sends them to the board. According to my testing that rate is over 100Hz. Shouldn't the xplane setup be sending messages at a lower rate so the comm link doesn't get flooded. Isn't this an xplane setup problem?

DonLakeFlyer, 2016-07-10:
I'm going to rate limit the HIL outputs since that will be an easy change.

DonLakeFlyer, 2017-12-29:
This seems to be fine now given the fact that folks are using HIL and not complaining.

closed 2017-12-29 by DonLakeFlyer