mavlink/qgroundcontrol #2556: PX4 - QGC setting quad - Power tab

lvale, 2016-01-08:
Setting up a Quad using PX4 master on the power tab there is no place to set up the battery capacity, that has to be set on the Parameters screen. I believe it makes sense to have the Battery capacity on the Power Tab.

[image]

[image]

DonLakeFlyer, 2016-01-10:
Although there is a BAT_CAPACITY parameter I'm not sure anything triggers off of that value? Hence, setting it does nothing interesting.

lvale, 2016-01-11:
Theoretically if the "reserve capacity" is reached then Failsafe gets active. And the reserve is calculated by Design capacity - consumption (mAh)= remainder > reserve cap

[image]

lvale, 2016-01-11:
Just checked the code and it's being verified here:

https://github.com/diydrones/ardupilot/blob/8fdbb36827ca489df157984c4a8b444a7300a6e4/libraries/AP_BattMonitor/AP_BattMonitor.cpp#L274

lvale, 2016-01-11:
Sorry. Wrong fw. I was thinking APM.

closed 2016-01-11 by lvale

DonLakeFlyer, 2016-01-11:
@LorenzMeier Is the bat capacity (maH) used anywhere on PX4 stack? Seems like I remember it wasn't yet.