mavlink/qgroundcontrol #2560: QGC default map location

lvale, 2016-01-09:
I believe that the default location shown on the map could be saved on the last location received by a vehicle instead of just jumping back to SF.

Example: I disconnected the vehicle while in Europe, and as soon I disconnected the location map just jumped back to the pic.....

[image]

lvale, 2016-01-15:
@DonLakeFlyer 
Sorted, thanks.
But :)

Vehicle had GPS lock and then was disconnected and this screen is correct, i.e. remained at the last known GPS Pos.

[image]

but if I move to the planning tab it goes back to SF

Shouldn't it inherit the GPS pos ??

[image]

DonLakeFlyer, 2016-01-15:
I can make it do that.

DonLakeFlyer, 2016-01-15:
It's doesn't inherit the GPS position, because that ui is only created when you click the tab. If you never clicked the tab, then it never saw the gps position go by because it wasn't running.

lvale, 2016-01-15:
I believe it makes sense to "inherit" the last known good position, in this case from the Fly tab.

DonLakeFlyer, 2016-01-17:
This is going to require some restructuring of the Fly View. The Map and Video items need to be persistent and just switch size and z-order. Otherwise you can't depend on the Fly view being around. By re-loading the items, the Fly view also also loses state like position I think, and re-creating all the mission and flight lines is expensive on the switch.

Other option is to leave things along and just push the map position out of the inner Map control up to the parent. This is doable as well. Likely easier, but still not sure if it's better the re-structure map/video setup.

@dogmaphobic Did you happen to try this structuring first and run into problems? Not sure yet if it's worth the effort to restructure map/video switching mechanism.

dogmaphobic, 2016-01-18:
> Did you happen to try this structuring first and run into problems? Not sure yet if it's worth the effort to restructure map/video switching mechanism.

I'm not even sure I understand what you are asking :)

The whole Map view UI code has changed so much since I first created, I can't barely recognize it. What is it that it is desired and it's not working?

DonLakeFlyer, 2016-01-18:
Currently the PIP and Map change positions by reloading themselves into loaders. So they are desrtroyed/re-created on the swap. There is a main view loader and a PIP loader. I'm wondering if instead it would be better to just have the two items always in existence and just change size/position/z-order when you want to swap. That would you have a persisten connection to the Fly view that you can use for other things. Which is what I need to be able to carry over Fly view map position into Plan view map.

dogmaphobic, 2016-01-19:
Tricky but should be doable. I will take a stab at it. I want to get *Over The Air* firmware update done first though...

DonLakeFlyer, 2016-01-21:
@dogmaphobic Sending to you for the Fly View change, then back to me the fix Plan view.

dogmaphobic, 2016-01-22:
Working on it...

closed 2016-02-13 by DonLakeFlyer