mavlink/qgroundcontrol #2561: Wrong VDOP displayed

lvale, 2016-01-09:
When connected to APM 3.3.2 copter this value is wrong (I'll have to check but I think is not sent by APM)

[image]

dogmaphobic, 2016-01-09:
That means the value was not present. The receiving code initializes the value to 1e10f (an absurd value) to indicate it's not present. The dialog missed that and it's displaying it any way. It should just say "N/A" (or something along those lines).

dogmaphobic, 2016-01-09:
@lvale Can you please check and see if this indeed resolved the issue?
Thanks!

lvale, 2016-01-10:
It's ok now. But spotted another issue. I'll open it

[image]

closed 2016-01-10 by lvale