mavlink/qgroundcontrol #2563: pixhawk latest firmware connection issue

blankered, 2016-01-09:
Hello,
I was using qgroundcontrol v2.8.0 to update my pixhawk to the latest firmware and found that it does not connect to qgroundcontrol after flashing it. It does however connect to qgroundcontrol after flashing the APM firmware. I am confused why this is happening, the ports in qgroundcontrol were the same when compared to the dmesg output in Ubuntu 14.04. Is this a firmware issue or qgroundcontrol issue?

LorenzMeier, 2016-01-09:
Can you please re-flash? I have cross tested this and it should work just fine with the latest code.

LorenzMeier, 2016-01-09:
It would be fantastic if you could re-try and provide us an update.

blankered, 2016-01-10:
I reflashed the pixhawk to the px4 firmware using v2.8.0. After I flashed it, I disconnected and reconnected the USB. I checked the dmesg output and verified the connection settings in QGroundControl. QGroundControl can find the px4 and flash it accordingly but wont bring up the parameters or calibration page. I have tried both the pre-compiled version and the qt compiled version with the same effect.

DonLakeFlyer, 2016-01-10:
> Error on link /dev/ttyACM0. Error connecting: Could not create port. Permission error while locking the device

This is an OS problem not a GGC problem.

blankered, 2016-01-10:
@DonLakeFlyer This only occurs when I flash the px4 firmware. GGC connects when I flash the apm firmware. Why would it be firmware specific?

LorenzMeier, 2016-01-10:
@blankered Its still an OS issue. It likely has nothing to do with Firmware but the sequence of your trial and error steps. Have you tried re-flashing the most recent PX4 Firmware (like from *now*) and trying again?

Also make sure to connect to the back of your PC (NOT A HUB) and to not power your whole drone off USB (your USB doesn't provide enough power)

DonLakeFlyer, 2016-01-10:
I can't answer that, but that error is exactly what you get when there is an OS permission issue. Given the fact that you are clicking a Connect button I assume you are running a Stable build. I'd also make sure you are using a daily build.

blankered, 2016-01-10:
I have both the stable build and daily build. I tried flashing the px4 firmware using the daily build, a usb port off the back of my tower, and battery pack connected. It still fails to connect after flashing the firmware, the pixhawk boots and starts to flash red rapidly. If I use QGC to flash the apm firmware then I have no problems getting the pixhawk to connect and show the summary page.

LorenzMeier, 2016-01-10:
Can you flash it again today, unless you have done that already? I want to be sure that both QGC and the Firmware are absolutely recent - depending on the combination they need to be absolutely recent.

blankered, 2016-01-10:
OK, so I reformatted the SD and it fixed itself. QGC has established a connection with the px4 after the flashing process. I reflashed using the daily build with the px4 master build firmware. I am going to do so testing to see if the problem is actually fixed and report back later. Thank you for your help.

LorenzMeier, 2016-01-10:
That's quite interesting. Could you try that at the end again to see if maybe flashing APM actually threw off the PX4 boot?

DonLakeFlyer, 2016-01-11:
@blankered What's the state of this? Is there a QGC problem here?

Flugsicherung, 2016-01-15:
Hello together. I have exact the same Problem as "blankered" I can´t connect to my Pixhawk after flashing the stable Firmware on QGC 2.8.0.  I am useing QGC on Windows 7 64 Bit. Flashing works only if the Pixhawk is NOT powered with external Power. Only if I flash with USB Power onIy can I flash it otherwise the flashing is not start up. On QGC 2.7.1 it works fine so I guess it must be a issue in QGC 2.8.0 or in last stable Firmware.

LorenzMeier, 2016-01-15:
It is designed to prevent reboots if the system is pre-armed, which probably happens when you connect the battery. So it all works as it should.

@DonLakeFlyer Do the instructions ask users to *ONLY* have USB connected already?

Flugsicherung, 2016-01-15:
I flash Firmware 1.1.0 and I can connect to the Pixhawk and the Summary Page is shown.
I make a Test and flash 1.1.1 again.....

Flugsicherung, 2016-01-15:
I flash Firmware 1.1.1 and same Problem, I can´t connect to Pixhawk.....

DonLakeFlyer, 2016-01-15:
> Do the instructions ask users to ONLY have USB connected already?

It doesn't say no battery. That wording is starting to read like a book! I'll change it. I think I'm going to change it to output after a failed flash as well.

Flugsicherung, 2016-01-15:
Sorry but why do you close this issue? The Problem is not to make the flash of the Firmware. Flashing without battery works fine. But if I flash Firmware 1.1.1 I can´t get a connection after successful flashing, but with Firmware 1.1.0 it works!!!!
Please can you explan it for me, thanks.

DonLakeFlyer, 2016-01-15:
Can you switch to using a QGC Daily Build. Stable is quite old and this point. If you still can't connect after that. See here for getting connection logging: http://www.qgroundcontrol.org/logging. Then put log output here.

LorenzMeier, 2016-01-15:
@Flugsicherung I'm checking the release. Let me make sure I didn't screw up and something bad slipped in.

Flugsicherung, 2016-01-16:
@DonLakeFlyer with Software do you mean? The Beta Version? If so my Virus Scanner (AVG) delete this File, he found a Virus inside. So I put the File to https://www.metascan-online.com the Scanner found a Virus too. 
[image]

@LorenzMeier Thanks, I thing the last "Stable" Versions of QGC (2.8.0) and PX4 Firmware (1.1.1) should play together. Normal Users takes only this Versions.

Flugsicherung, 2016-01-16:
Today I flash the Master Version of the PX4 Firmware but same happens, I can´t connect to the Pixhawk with QGC 2.8.0 flashing works fine. Then I flash the Version 1.1.0 back to the Pixhawk and I can connect without Problems.

DonLakeFlyer, 2016-01-16:
It is extremely doubtful that qgroundcontrol has a virus in it. This is a false positive. You should report the false positive to you virus folks.

DonLakeFlyer, 2016-01-16:
[image]

DonLakeFlyer, 2016-01-16:
I'll deal with reporting the false positive.

DonLakeFlyer, 2016-01-21:
Is this still a problem?

LorenzMeier, 2016-01-21:
Should be fixed by a Firmware fix and the SD card warning fix.

closed 2016-01-21 by LorenzMeier

Flugsicherung, 2016-01-22:
Ok I wait for the next Firmware, thanks.

LorenzMeier, 2016-01-22:
You don't have to wait. Just flash master (see the checkbox on the bottom of the flash dialog for "advanced options").

salim616, 2016-12-12:
Hi  @LorenzMeier 
I got the same problem as above. can not connect to my Pixhawk. i am V3.0.1
[image]
any help ?

msadowski, 2017-04-10:
I just run into the same problems as described in this thread (there was APM previously installed on my board). The Px4 would flash through QGroundControl and using make in the repository but it would not boot properly (QGroundControl didn't show anything). I tried latest release, latest commit from master branch and even some old commit from February. 

As @blankered noticed formatting the SD card helped and now Pixhawk boots properly.

DonLakeFlyer, 2017-04-10:
This problem doesn't have anything to do with QGC. If there is a problem it is  firmware problem.

feemi, 2019-07-04:
Hi @LorenzMeier @DonLakeFlyer,

This issue was closed long time ago. Could you find the root cause of the problem ?

We are having exact issue with various versions including v1.9. We are having this problem in mac and windows but not in ubuntu with pixhawk 2.1 cube fmu-v3 board.

pixhack5+ fmu-v5 is working properly.

Steps to reproduce:

1. If i reflash px4 firmware nothing changes. 
2. If i flash to ardupilot it connects successfully to qgc.
3. Then i reflash px4 it also connects successfully to qgc.
4. But if i select airframe or change any parameter and restart then the issue raises again.

Note: This problem only occurs with usb connection. It connects successfully from telem1 uart via ftdi or similar.

DonLakeFlyer, 2019-07-04:
Can you enter a new issue with exact details. There are multiple things dicussed in this issue. So what the "exact issue" is unclear.