mavlink/qgroundcontrol #2567: Disable input / output processing

LorenzMeier, 2016-01-10:
We should ensure we have this enabled on the UART:

https://github.com/mavlink/c_uart_interface_example/blob/master/serial_port.cpp#L359-L495

I can see that the parameter count gets modified by the input processing on my machine.

LorenzMeier, 2016-01-10:
With this debug diff:

```diff
diff --git a/src/uas/UAS.cc b/src/uas/UAS.cc
index 7b2c03f..d3466fa 100644
--- a/src/uas/UAS.cc
+++ b/src/uas/UAS.cc
@@ -774,6 +774,11 @@ void UAS::receiveMessage(mavlink_message_t message)
         {
             mavlink_param_value_t rawValue;
             mavlink_msg_param_value_decode(&message, &rawValue);
+            if (rawValue.param_count != 386) {
+                qDebug() << "PARAM VAL DECODE FAIL!!!";
+                qDebug() << "param count: " << rawValue.param_count;
+            }
+            rawValue.param_count = 386;
             QByteArray bytes(rawValue.param_id, MAVLINK_MSG_PARAM_VALUE_FIELD_PARAM_ID_LEN);
             // Construct a string stopping at the first NUL (0) character, else copy the whole
             // byte array (max MAVLINK_MSG_PARAM_VALUE_FIELD_PARAM_ID_LEN, so safe)
@@ -1435,6 +1440,12 @@ void UAS::processParamValueMsg(mavlink_message_t& msg, const QString& paramName,
 
     qCDebug(UASLog) << "Received PARAM_VALUE" << paramName << paramValue << rawValue.param_type;
 
+    qDebug() << "received param:" << rawValue.param_index << "count:" << rawValue.param_count;
+
+    if (rawValue.param_count != 386) {
+        qDebug() << "+++++++++++++++++++++++++++++++++++++++++++++++++";
+    }
+
     emit parameterUpdate(uasId, compId, paramName, rawValue.param_count, rawValue.param_index, rawValue.param_type, paramValue);
 }
 

```

I get this string feedback on a param load attempt on PX4 master via radio:
```
received param: 11 count: 386
PARAM VAL DECODE FAIL!!!
param count:  246
received param: 11 count: 386
received param: 21 count: 386
```
Note that we're receiving it twice: Once with the correct payload data, once with the incorrect count. I have no idea what's going on there.

LorenzMeier, 2016-01-10:
I debugged this on the Firmware side. Its sending the right thing!

LorenzMeier, 2016-01-10:
So the data was indeed garbled. Because of that frickin 2nd Pixhawk setup I had sitting on the CI hardware system which had the same radio ID. Cost me a complete afternoon. I hate it!

closed 2016-01-10 by LorenzMeier

DonLakeFlyer, 2016-01-10:
Are you sure that doesn't have something to do with #2569 as well? Haven't tried to repro yet, but I was running radio last week without issue on OSX.