mavlink/qgroundcontrol #2574: Configurable units in the instruments / flight view

julianoes, 2016-01-11:
It would be nice if the units of the flight view could be changed, for instance I'd want to use m/s instead of km/h.

If this has already been added after 2.8.0, excuse my ignorance!

DonLakeFlyer, 2016-01-11:
This is doable but first the firmware needs to standardize on parameter unit definitions. Right now most parameter units are just unknown quality strings. So you can't really trust that "km/h" is what you expect in order to translate it. It requires validation of units on the firmware parameter meta data generation side as well as validation against a known set.

julianoes, 2016-01-11:
Ok but the units are mostly defined in the mavlink comments, right?

DonLakeFlyer, 2016-01-11:
Ah, sorry you said "Flight View". Those are more locked down and don't need parameter meta data. That could all be done on the QGC side.

julianoes, 2016-01-11:
Yes, I guess that's what I meant.

julianoes, 2016-01-12:
FYI @georgehines

DonLakeFlyer, 2016-01-21:
Need to consider whether for now we change to m/s and do confgurability in 3.1

DonLakeFlyer, 2016-01-21:
@lorenz Feedback?

dagar, 2016-03-02:
Despite my best efforts airspeed in knots and height in feet is going to be important for us.

DonLakeFlyer, 2016-03-02:
This is easier to do now that instrument panel is using the FactGroup stuff.

closed 2016-03-13 by DonLakeFlyer