mavlink/qgroundcontrol #2586: ParameterlLoader handling of multiple components and hash

DonLakeFlyer, 2016-01-12:
- Request parameter list as well as hash values at start
- The hash cache must be separated out by component
- hash values will come back from each component
- If has comes back from component and the hash matches write a value back to that components hash value, which will in turn stop the param list. Then use the cache for that component

dogmaphobic, 2016-02-01:
While at it, this:

```
 /* compute the crc of the local cache to check against the remote */
    foreach(int component, cache_map.keys()) {
        foreach(int id, cache_map[component].keys()) {
            const QString name(cache_map[component][id].first);
            const void *vdat = cache_map[component][id].second.second.constData();
            crc32_value = QGC::crc32((const uint8_t *)qPrintable(name), name.length(),  crc32_value);
            crc32_value = QGC::crc32((const uint8_t *)vdat, sizeof(uint32_t), crc32_value);
        }
    }
```

makes an assumption all values are sizeof(uint32_t), which assumes the memory location where they were first set was initialized (zeroed) before setting. I had the CRC running only for the length of the actual **type** and was getting different values than those computed by QGC.

DonLakeFlyer, 2016-02-10:
> write a value back to that components hash value, which will in turn stop the param list.

@LorenzMeier Has this change been made on the firmware side?

LorenzMeier, 2016-02-10:
not yet

NaterGator, 2016-02-10:
> While at it, this: [...] makes an assumption all values are sizeof(uint32_t), which assumes the memory location where they were first set was initialized (zeroed) before setting. I had the CRC running only for the length of the actual type and was getting different values than those computed by QGC.

To be fair I made that assumption knowing the only two union initializers used in the python script are float and 32-bit int:
https://github.com/PX4/Firmware/blob/56d67c6f6c8f19d5362936257ed4e6a76513c0dc/Tools/px_generate_params.py#L76

DonLakeFlyer, 2016-02-10:
@NaterGator You familiar with this code, wondering if you could update it:
- To follow the new spec
- Also need to handle type size appropriately so that we don't need to rely an extra space being zeroed. ArduPilot uses additional type sizes smaller than uiint32_t, so we'll need that to bring them on board. Also to guard against PX4 adding additional smaller types further down the road.

NaterGator, 2016-02-10:
Sure I can spend some time on this. Is the "new spec" [this](https://github.com/mavlink/qgroundcontrol/issues/2586#issue-126235663)? Should we plan to maintain a separate caches for each system ID  + component?

DonLakeFlyer, 2016-02-10:
Correct on spec. Yes I think cache for each vehicle as well makes sense.

dogmaphobic, 2016-02-10:
> Also need to handle type size appropriately so that we don't need to rely on extra space being zeroed.

I used to compute the CRC based on the value type (sizeof(type)). I've changed the code in the ESP firmware to copy the value into a zeroed out uint32_t to match QGC's code.

https://github.com/dogmaphobic/mavesp8266/blob/master/src/mavesp8266_parameters.cpp#L172-L186

If we switch to sizeof(type), I need to revert it. The PX4 firmware uses sizeof(val). We just need to make it consistent across all sides:

https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param.c#L1109-L1130

NaterGator, 2016-02-10:
I'm all for fixing it to only hash enough bytes to completely cover the storage type specified (less overhead work too). 

Technically the CRC32 implementation in NuttX is not spec-compliant since it doesn't properly handle pre/post conditions, but most CRC32 implementations I've found are not spec compliant so I'm inclined to leave it be.

LorenzMeier, 2016-02-17:
Should I be merging the the Firmware PR?

NaterGator, 2016-02-17:
I don't have the ability to test ESP against this, @DonLakeFlyer can you take a swing at that before we merge these PRs?

DonLakeFlyer, 2016-02-17:
@dogmaphobic That would be Gus not me. I don't have one of those things.

LorenzMeier, 2016-02-17:
right, let me test that quickly

LorenzMeier, 2016-02-17:
Yep - just downloaded a 3MB log off Pixracer via Wifi. Need to do traffic shaping onboard, but other than that looking good.

DonLakeFlyer, 2016-05-13:
This is all working correctly now isn't it?

LorenzMeier, 2016-05-13:
I think so.

DonLakeFlyer, 2016-05-13:
Closing. Re-open if this is incorrect

closed 2016-05-13 by DonLakeFlyer