mavlink/qgroundcontrol #2593: PX4Flow video downlinking tab missing

harewei, 2016-01-13:
Hello

I've just installed qgroundcontrol by pulling from
git clone --recursive https://github.com/mavlink/qgroundcontrol.git
and compiling with Qt5.5.1

I'm able to run it, but I could no longer find the PX4Flow video streaming tab that existed a couple of days ago.  Could someone tell me where it is?  If not, how do I install the old version, such as 2.7.1?  Because I tried downloading the source.zip file and compiling with Qt, but it says error need to build from shadow copy.

Thanks.

DonLakeFlyer, 2016-01-13:
The PX4Flow has to be connected directly to QGC via USB for it to show up. Nothing has changed here in a while.

harewei, 2016-01-13:
I connected with USB, but the it was showing up as if a coptor has been connected (with the airframe, radio setup etc tabs on the left hand side), but no video downlink tab

DonLakeFlyer, 2016-01-13:
I’ll try it tomorrow to see what I get.

harewei, 2016-01-13:
ok.  oh and by the way, the 2.8 version which I could see video downlink was from a week ago, and from memory it had a qgroundcontrol-splash.sh which I called directly, without me needing to compile through qt.  

I have since then reformatted my laptop and downloaded the new v2.8.0-483-g438cf36 (development) that I needed to compile with qt, and launched with 
~/build-qgroundcontrol-Desktop_Qt_5_5_1_GCC_64bit-Debug/debug$ ./qgroundcontrol

DonLakeFlyer, 2016-01-13:
PX4Flow is working fine for me on Daily Build:
[image]

If you are seeing setup buttons like Radio Config and so forth then are you sure you are not connecting a Pixhawk via USB which has a PX4Flow then connected to it. In order to get video setup button to focus the lens you must connect PX4Flow directly to QGC, no Pixhawk board.

harewei, 2016-01-14:
Ok I figured out the problem.  I was using px4flow-klt-06Dec2014.px4 (which is the one I used to get loiter working with optical flow) for px4flow firmware before, and with that version installed, the video did not show up in QGC.

So the steps to solving my problem was, flashing with the latest stable PX4Flow firmware, change the lens focus with video display, then flash the custom firmware.

Thanks for the help.

closed 2016-01-14 by DonLakeFlyer