mavlink/qgroundcontrol #2599: Airframe Config Icon discussion

DonLakeFlyer, 2016-01-13:
Issues:
- There is only one set of airframe icons for QGC that encompass both PX4 and APM. No need to split them out into two sets. A QuadRotor Plus icon for example is the same for both PX4 and APM.

Give me a day or so to determine the full set of issues. I'll let you know when I'm done with the set then we can discuss as needed.

cfelipesouza, 2016-01-13:
Great @DonLakeFlyer! Just to clarify, I split into two sets just to organize myself and the file I sent you. I totally agree that both should use the same icons. :)

DonLakeFlyer, 2016-01-13:
Issues with current set:
- There is only one set of airframe icons for QGC that encompass both PX4 and APM. No need to split them out into two sets. A QuadRotor Plus icon for example is the same for both PX4 and APM.
- I think the Helicopter should use the same color circle rotor visualization as the others (instead of showing individual blades).
[image]

DonLakeFlyer, 2016-01-13:
@cfelipesouza First set of comments are good to go. I'm working to integrate the current set into the codebase now.

DonLakeFlyer, 2016-01-13:
@LorenzMeier For the Coaxial setups do those have Plus and X variants?

LorenzMeier, 2016-01-13:
We have Coaxial in + and x layout, yes. Although I think the + layout is not flown by anyone any more these days.

a foo coaxial has foo / 2 rotors on top and bottom each. So a coaxial octorotor has 8 rotors.

DonLakeFlyer, 2016-01-13:
Ok, so we need Coaxial Icons in both style. I'll update the icon list.

dogmaphobic, 2016-01-13:
> Ok, so we need Coaxial Icons in both style. I'll update the icon list.

I assume only for Octos, not Hexa (Y config)?

DonLakeFlyer, 2016-01-13:
Removed Hexa Coxial from list since these are Y6A and Y6B variations.

DonLakeFlyer, 2016-01-13:
Changed TriCopter naming to YPlus/YMinus to match Y6A/Y6B naming.

dogmaphobic, 2016-01-13:
Isn’t a tri Copter one where the tail has a servo to tilt it? In that case, Y6A/Y6B are Hexa Coaxials, right?

DonLakeFlyer, 2016-01-13:
TriCopter is three blade Y configuration (PX4/APM seem to use this naming) and Y6* for Hexa Coaxial.

For a Hexa Coaxial in PX4 terminology, which one is it:
[image]
[image]

DonLakeFlyer, 2016-01-13:
Drawing bug, Heli is png, Flying Wing is svg:
[image]

cfelipesouza, 2016-01-15:
Here is the new icon set. Waiting for comments before export. :)

[image]

DonLakeFlyer, 2016-01-15:
@LorenzMeier Is this one correct (it's not supposed to be coaxial):
[image]

DonLakeFlyer, 2016-01-15:
Oops, Also does one work in Light theme?
[image]

cfelipesouza, 2016-01-15:
For sure.
Light Theme color background is #d9d9d9

[image]

LorenzMeier, 2016-01-15:
@DonLakeFlyer The quad tailsitter is ok.

DonLakeFlyer, 2016-01-15:
Awesome we're good to fo then. Thanks @cfelipesouza

DonLakeFlyer, 2016-01-21:
Ok, just tested this and we can close this now. Thanks everyone.

Found two bugs:
-  #2686
- https://github.com/PX4/Firmware/issues/3562

closed 2016-01-21 by DonLakeFlyer