mavlink/qgroundcontrol #2604: New Aiframe Config Proposal

DonLakeFlyer, 2016-01-13:
[image]

Here is a completely hacked up minimalist screen shot of the new airframe config proposal. The basic concepts:
- Top of screen shows current airframe selection name
- There should be some more text fields below that like: Maintainer and probably a longer text description. Maybe others. TBD.
- Clicking Change Aiframe opens a right side slight out dialog which allows you to pick a new airframe. There is where the new airframe picker UI design should go.
- If the airframe is currently not set, the airframe picker dialog should open automatically.
- The lower portion of the config dialog shows a detailed prop rotation and motor number visual. I plan to be able to generate these automatically from the airframe xml.
- At the very bottom is a visual for wiring help which shows you what wire plugs into what port on the Pixhawk.
- Need to stuff some other help type type in places to explain things.

DonLakeFlyer, 2016-01-13:
@biapalmeiro @cfelipesouza Here is the new airframe flow with hack visuals as discussed in #2440

biapalmeiro, 2016-01-14:
@DonLakeFlyer Thanks. We'll take a look and give some feedback soon.