mavlink/qgroundcontrol #2605: MAVLink secure link icon

LorenzMeier, 2016-01-13:
We would want to create a new icon to indicate a secure link, much like HTTPS in a web browser.

LorenzMeier, 2016-01-13:
@biapalmeiro @cfelipesouza Essentially the equivalent of the padlock in the browser.

cfelipesouza, 2016-01-14:
Where this icon will be placed??

LorenzMeier, 2016-01-14:
In the status bar at the top to indicate the user wether he has a secure link.

dogmaphobic, 2016-01-14:
I would put it up on top on the toolbar. Something like this (showing an insecure connection). Clicking/touching the icon would bring up a small pop up explaining to the user.

[image]

LorenzMeier, 2016-01-14:
It would be fantastic if it could be a MAVLink / drone specific style. So that by looking at the icon, you know its secure MAVLink (and not just some "lock").

cfelipesouza, 2016-01-18:
What do you guys think?

[image]
[image]

DonLakeFlyer, 2016-01-18:
The issue is that it is QuadCopter oriented. People get consfued when we do things like that and they are flying a Plane or piloting a Rover or a Sub.

LorenzMeier, 2016-01-19:
Yes, second that. It should not include the airframe itself. And it should be visually nimble so that you recognise it, but at the same time doesn't eat a lot of screen real estate.

biapalmeiro, 2016-01-19:
Just to understand. Is it to show the drone is connected or it is an extra information?

LorenzMeier, 2016-01-19:
It's the equivalent of the padlock in your browser to indicate a secure connection.

cfelipesouza, 2016-01-19:
I understand that the icon represents a secure connection, but for us here, the information about the drone connection is missing, like on Mission Planner, where it has a socket to represent it. And it is more important, what do you think? Do you guys have the intention to implement it? Maybe we can merge this information (connection + security) in one single icon.

Another question: the scenario where the drone is connected but the connection is not safe. Does it happen?

LorenzMeier, 2016-01-19:
Compare the connected and unconnected toolbar. The connected state is obvious.

biapalmeiro, 2016-01-19:
@LorenzMeier We can notice the difference, but maybe for new users it is not enough. Or maybe if the entire screen changed and redirect the user to the correct screen could solve this perception. What do you think?

DonLakeFlyer, 2016-01-19:
There is quite a bit of UI change that happens when a vehicle connects:
- Toolbar changes radically
- Parameter loader progress bar goes across the top of the screen
- Vehicle shows up on map
- Setup View populates with buttons

We haven't had anyone confused as to when something was connected. Don't think we need to add something. Less is more. I also think that the lock icon or whatever shouldn't be on the main screen. Does that need to take up space the entire time the vehicle is connected. After you see it once it provides no additional information. Vehicle Summary would be a better place. Possibly including a bottom popup, that tells you once you are on a secure connection.

biapalmeiro, 2016-01-19:
@DonLakeFlyer Agreed the Vehicle Summary is the best place for that. In this case we don't need an icon, once this screen shows only text information. Neither a pop up, because all the info could be displayed directly on this screen. It could be placed on Safety block.

closed 2026-10-04 by github-actions[bot]