mavlink/qgroundcontrol #2614: Usability, scrolling lists

lvale, 2016-01-15:
On APM parameters, the scrolling list of parameters when too long and scrolled to the bottom, when changing to a smaller list does not appear.

Example:
Check this:
[image]

scroll down to the bottom of this list:

[image]

now move to a smaller list of parameters like LAND, and the list is apparently empty

[image]

but no, it's hidden at the top....

[image]

Also there is no user feedback if the list is greater than one screen, so you'll have to try to scroll it to see if something is there, either up or down.

DonLakeFlyer, 2016-01-15:
Related to #2469 which I just moved to 3.0.

DonLakeFlyer, 2016-01-15:
Tracking scroll feedback in #2649

closed 2016-01-15 by DonLakeFlyer

lvale, 2016-01-15:
@DonLakeFlyer In this case it might be something else. I believe that when a new item is selected on the left list it should "reinitialize" the position of the list to be at the top, so that when one changes from category at the left, the list is always "painted" from the beginning.

DonLakeFlyer, 2016-01-15:
I know. I fixed that in #2618