mavlink/qgroundcontrol #2628: Support custom download / web server URL

LorenzMeier, 2016-01-15:
@DonLakeFlyer When working with non-developer early adopters we start to run into cases where we would need to provide them auto-built binaries but off custom builds. We need to generalise the download locations for PX4. Any particular suggestions?

LorenzMeier, 2016-01-15:
Urls for PX4 on Pixhawk: https://github.com/mavlink/qgroundcontrol/blob/master/src/VehicleSetup/FirmwareUpgradeController.cc#L216-L218

AndreasAntener, 2016-01-15:
Suggestion for startup parameters (or something similar), to narrow down the update choices:
- base URL
- types (apm, px4, custom binary)
- which of the 3 types should be available (stable, beta, dev)

DonLakeFlyer, 2016-01-15:
> provide them auto-built binaries but off custom builds.

Two versions of Custom:
- Custom (local file)
- Custom (download) prompt for url

DonLakeFlyer, 2016-01-16:
Curious as to why you would need something built into QGC for this. Seems like it just add complexity to things. Why not:
- Email them build
- or Provide then a url to the build, they then click that in a browser which downloads it to there machine
- Then they just use Custom to install?

DonLakeFlyer, 2016-01-21:
I'm planning to move this to 3.1 release. Speak up now if you disagree.

closed 2026-10-04 by github-actions[bot]